letscontrolit/ESPEasy #940: Can not read disable dummy device value

wolverinevn, 2018-02-25:
### Steps to reproduce
- OTA update to **mega-20180224** from **2.0.0-dev12**
- In rules: use **if** statement compare **disable dummy** device value to any value. Ex:
if [Dummy#value1] > 1000
command
endif
* Taskvalueset works well.
Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
- Yes

### Expected behavior
- Read disable dummy device value as normal sensor value
### Actual behavior
- Can not read value of disable dummy device
### System configuration

Hardware: Nodemcu-12E

Software or git version: mega-20180224
Previous good version: 2.0.0-dev12

uzi18, 2018-02-25:
It is not a bug.
Why do You want to read from disabled device?

TD-er, 2018-02-25:
It is indeed by design to ignore disabled devices/controllers etc.
This will save resources on the ESP and allow for better responsiveness and more stable operations.

wolverinevn, 2018-02-25:
I use dummy devices as variables so no need to keep it updating regularly (in log: EVENT: ...). I want to control them manually by reading and writing as needed. I do not know well how ESPeasy works but I guest keeping device disabled helps saving resource.

closed 2018-02-25 by wolverinevn

TD-er, 2018-02-25:
I guess enabling the dummy device will not use many resources, so please try with the dummy device enabled.