letscontrolit/ESPEasy #945: v20 compile error - ESP8266Ping.h missing

s0170071, 2018-02-25:
mega compiles fine but v2.0 is missing ESP8266Ping.h

/home/john/Arduino/scetchbooks/ESPEasy/ESPEasy.ino:374:25: fatal error: ESP8266Ping.h: No such file or directory
 #include <ESP8266Ping.h>

TD-er, 2018-02-25:
Apparently you are not compiling using PlatformIO???
The module is marked in the ini file as dependency.
It is a standard Arduino module you can install using the module manager.

If not, please let me know, then I will add it in the distributed lib folder

s0170071, 2018-02-25:
No, I use Arduino IDE 1.8.5
Please add it to the lib folder as others will run into that issue as well.

TD-er, 2018-02-25:
Will do!

closed 2018-02-26 by s0170071

uzi18, 2018-02-26:
@s0170071 I'm using modified version of Sloeber (eclipse with arduino plugin) to build ESPEasy.
PlatformIO is not working on 32bits.
Arduino IDE is not suitable for such big project.