letscontrolit/ESPEasy #950: Ping plugin

Grovkillen, 2018-02-25:
I'd love to have one unit pinging some outside machine/server and if the result is above a certain value it'll alternate a relay for my router making it reboot.

I guess the ping result of a non - responsive unit should be set to "-1" for the sake of the logic test.

Another use case could be to ping my phone/car and if no response I have left my house and thus do something with that info.

TD-er, 2018-02-25:
Ah, someone has been peeking into the new sources ;)

Grovkillen, 2018-02-25:
Haha!! :)

Grovkillen, 2018-02-25:
Can't we do something like this as well 😋🤤

https://github.com/kalanda/esp8266-sniffer

I need that for a curling project of mine 😉 Equipping the stones with esp-01 units only broadcasting and have multiple esp units around the ice reporting the stones rssi values and then triangulating the stones position. (Major feature! 😁)

Grovkillen, 2018-04-01:
http://www.letscontrolit.com/forum/viewtopic.php?t=3636

Here's a old plugin that might be needing a bit of a remake and then get to be a official plugin?

Grovkillen, 2018-10-24:
I close this one, continue in this thread:  #1863

closed 2018-10-24 by Grovkillen