letscontrolit/ESPEasy #951: FR: New event to process rules at startup - before network init

M0ebiu5, 2018-02-25:
The event "System#boot" now fires after wifi, network and mqtt are initialized. This takes many seconds and prevents a responsive power up behavior (eg: a lamp).

I would propose to initialize the plugins before the network (if there are no side effects) and to introduce a new event, which fires before all the network stuff. 

Forum thread:
[https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=4902](url)

Grovkillen, 2018-02-26:
System#Initialized perhaps.

uzi18, 2018-02-26:
When looked into setup it is not possible to launch this very early. 
We initialize plugins realy late and now wifi. 

Need to think also about system#connected,  System#disconnected events to add.

TD-er, 2018-02-26:
This is a good idea, but I think we should first make a list of events, like the two @uzi18 mentioned already.

uzi18, 2018-02-26:
If we add event connected/disconnected we can try to add this to plugins api ...

uzi18, 2018-02-26:
Looks like right place.
In my opinion we cant move Boot event there for support all versions before and System@initialized suites here better.

M0ebiu5, 2018-02-26:
I moved the boot event just to try out, if it works.
The official version should use a new event, so it wouldn't break current behavior.

TD-er, 2018-03-09:
Could you test the upcoming build, to see if it now has enough options to get this working?
See #1032

Grovkillen, 2018-03-16:
I want to close this. Tested @M0ebiu5 ?

closed 2018-03-17 by Grovkillen