letscontrolit/ESPEasy #952: Some cosmetics when compiling with the Arduino IDE

micropet, 2018-02-25:
The category in the three libs is wrong. (CHT16K33 MechInputs SerialSensors)
And for LiquidCrystal_I2C please set the right platform.

WARNUNG: Kategorie 'LED' in der Bibliothek CHT16K33 LED Matrix Library ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie 'Input' in der Bibliothek MechInputs ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie 'Input' in der Bibliothek SerialSensors ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Bibliothek LiquidCrystal_I2C behauptet auf (avr) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp8266) Architektur(en) ausgeführt wird.

micropet, 2018-02-26:
I know.
But it is an unnecessary warning.

Grovkillen, 2018-10-23:
Since we're using Platformio ourselves we close this issue, it's not really related to the FW. Please open if I'm wrong.

closed 2018-10-23 by Grovkillen