letscontrolit/ESPEasy #953: Web interface very late available.

micropet, 2018-02-25:
With the current Mega Source it takes about 2 minutes to reach the web interface after booting.

TD-er, 2018-02-25:
I have noticed the same and I think it has to do with one of the last commits added.
In the last few days I have been running with the new webserver and these builds did not have so much delay after boot.

micropet, 2018-02-25:
OK,
I thought my Wemos D1 is broken and wanted to swap it - suddenly the web interface was back.

TD-er, 2018-02-26:
You could try the following:
Try to strip out the changes of this PR: #947
And perhaps the reserve line I added should be removed: https://github.com/letscontrolit/ESPEasy/blob/cbdddf8c064e4da729125449110dfea5a2cc9fd5/src/WebServer.ino#L38

s0170071, 2018-02-26:
Mine takes exaclty 3.31s from reset to page load. With an **empty configuration**, latest 2.4.x git and latest mega.
Last time the web interface has hung, it was related to MQTT...

TD-er, 2018-02-26:
MQTT should now take less resources to connect.
But I will look into it this evening.

Things I changed the last few days:

- MQTT connection retry is less resource hungry. (more dynamic scheduler, less retries at once)
- NTP does a lot more relaxed retry.
- A lot less often a DNS lookup. (DNS lookup will stall the ESP when it will fail)
- Added host online check, which uses ping (might be some timeout here?)
- The reserve line, I already linked above. (may result in more memory use than needed)

micropet, 2018-02-26:
With only:
 buf.reserve(BufferSize+100);
 removed it starts within 3 seconds.

s0170071, 2018-02-26:
@micropet: how much free ram do you have ? And whats the figure in ( ) ?

micropet, 2018-02-26:
Free Mem: | 12000 (10336 - sendContentBlocking)

s0170071, 2018-02-26:
thats more than enough. I don't see any connection to the buf.reserve(BufferSize+100); line...

micropet, 2018-02-26:
the buf.reserve(BufferSize+100); is now also removed.

TD-er, 2018-02-26:
When you append a lot, then there will be a lot of memory overhead.
So I will remove it this evening.
Misjudgement on my side :(

s0170071, 2018-02-26:
@micropet: the buf.reserve line isn't everything that you changed , is it?
@TD-er hes got 10k ram free and the latest commit should give a low memory warning in the web interface... The buf.reserve was put there by me, wasn't it ?

TD-er, 2018-02-26:
Nope, the one in the constructor without parameter was done by you, the one in the constructor with a String parameter was mine.
And apparently you use the one with String a lot as temporary and the one without parameter as a base to create a larger string.

s0170071, 2018-02-26:
Ah, now I see the second buf.reserve(). still, no problems with my ESP.
Whats that   
```
  StreamingBuffer(String &a) :
    initialRam(0), beforeTXRam(0), duringTXRam(0), finalRam(0), maxCoreUsage(0),
    maxServerUsage(0), BufferSize(400), sentBytes(0) {
      buf.reserve(BufferSize+100);
      buf = a;
    }
```
doing anyway ? StreamingBuffer exists exactly once 
`}TXBuffer;`
and only uses the default constructor. At least it compiles and runs without the code above.

TD-er, 2018-02-26:
Perhaps some left over code?
There are also += and + operators for a StreamingBuffer.
So maybe also look into those?  
And there may be only one TXBuffer, but that doesn't mean you cannot create other instances of the StreamingBuffer.

s0170071, 2018-02-26:
+= and + operators for a StreamingBuffer are definately leftover. But if that whole constructor is leftover it can hardly cause the webpage delays.

TD-er, 2018-02-26:
I agree and also have no explanation.
Perhaps one of the assignments with + / += operator and a String is actually using that constructor if defined?
As a test you could increase some internal counter in StreamingBuffer and see how often it is called?
Use a static one, to count all instances.

s0170071, 2018-02-26:
Tried that. The () constructor is called exactly one, the other one not at all.

TD-er, 2018-02-26:
The I would suggest to remove what's not being used.
So all operator definitions with StreamingBuffer as a type can be removed.
Just to be sure, define a copyconstructor and make it private. You don't want to copy it, just pass it by reference.

s0170071, 2018-02-26:
Like #958 you mean?

micropet, 2018-02-26:
@micropet: the buf.reserve line isn't everything that you changed , is it?

yes, thats all.

closed 2018-03-09 by TD-er