letscontrolit/ESPEasy #956: Add Telegram Bot to ESPEasy

TD-er, 2018-02-26:
@mstjerna suggested this as PR #955 , but it should be an issue:

> Hi
> 
> I found a great project on Git-hub that would make sense to implement into the ESP-Easy.
> https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/
> 
> It can be used for both notification and controll of I/O.
> It would be great to have it in the ESP Easy for easy reporting of temperature/uptime etc without having to go through OpenHAB or similar home automations...
> 
> Sorry for this, I opened this in the wrong section. :-(
> New feature request...
> 
> Br
> Michael Stjerna

uzi18, 2018-02-26:
It use WifiSecureClient I'm not sure if we got free RAM for SSL connection.

TD-er, 2018-02-26:
Also SSL is 2.4.0, so future plans.

Maybe we could also add some SSL gateway dependency on something like a raspberry py. That's also needed for most email providers and will become more of an issue in the near future.

closed 2019-10-11 by Grovkillen