letscontrolit/ESPEasy #957: Debug feature: use empty / restore configuration

s0170071, 2018-02-26:
I noticed that a lot of issues seem related to plugins or faulty configurations. They then affect the webpage generation or other stuff. It would be sensible to ask users to try an empty configuration if things are mysteriously wrong. 

I propose to add two commands: 
- ?cmd=backupandemptyconifg backups the current configuration (just rename), create a blank one and reboot.
- ?cmd=restoreconfig reverses / restores everything.

TD-er, 2018-02-26:
A bit like this suggestion: https://github.com/letscontrolit/ESPEasy/issues/861 ?

s0170071, 2018-02-26:
not the same but not too far away. #861 aims at faulty configurations (interrupted while writing etc) while this one allows to temporarily ignore most bugs introduced in other places, e.g. plugins

tonhuisman, 2023-09-03:
This seems to be a duplicate, so can be closed.

closed 2023-09-04 by TD-er