letscontrolit/ESPEasy #959: Implement mDNS to access device without  knowing the IP-Address

mstjerna, 2018-02-26:
Implement mDNS to access device without  knowing the IP-Address on the local network.
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS

This would ease up the WiFi-boarding process for users that are uncomfortable in accessing their Router to find the IP or use Serial debug to find it out...
On first boot the name would be http://esp_easy_0.local and then the users preferences when naming the device from the web-gui.

Br
Michael Stjerna

TD-er, 2018-02-26:
The IP address is already shown in the webinterface when configuring the WiFi for the first time.
Maybe we could let them already enter a hostname in that interface, combined with this addition?

mstjerna, 2018-02-26:
Yes that would be even nicer...
 
My experience is that I never see the WiFi IP since my phone disconnect from the ESP-AP when onbording the WiFi. The device is then "lost" and I need to in to the router to find the IP to reach the web-interface for Initial configuration. 

The mDNS would solve this and giving the possibility to give the device name on AP-onboarding would be even nicer.
Other solutions like the "Espruna" uses this technique.
https://github.com/xoseperez/espurna

They also provide a double click on the "Program button" to access the wifi configuration even if the device can attach to the wifi.

Grovkillen, 2020-01-10:
This one is implemented

a-x-, 2020-05-03:
This does not work

router says: 
Active Host Name: `living-light-ESP-Easy`
But http://living-light-esp-easy.local/ does not work locally

TD-er, 2020-05-03:
Tested it here on a recent build and indeed it does not resolve to a host.

Strange as it used to work.

uzi18, 2020-05-03:
is it related to : https://github.com/esp8266/Arduino/issues/7262

rvt, 2020-05-25:
can we not just add mDNS?

TD-er, 2020-05-25:
> can we not just add mDNS?

That is present, but apparently it is broken in the more recent esp8266/Arduino core releases.

rvt, 2020-05-26:
> > can we not just add mDNS?
> 
> That is present, but apparently it is broken in the more recent esp8266/Arduino core releases.

That is odd, I compile with latest esp8266/Arduino for other projects and mDNS is working fine. Just not with ESPEasy. Can somebody confirm that swell?

TD-er, 2020-05-26:
@rvt Have you enabled it in your build definition?
```c++
#define FEATURE_MDNS
```

tonhuisman, 2023-09-02:
No response, and no not-working issue with mDNS (though maybe other issues...), so this issue can be closed.

closed 2023-09-02 by TD-er