letscontrolit/ESPEasy #964: time contitions doesnt work - compare is only on hours.

uzi18, 2018-02-26:
Here You are:
Rules:   

```
on oldtime do
if %systime%>12:34
event ok
else
event nok
endif
endon

on newtime do
if %systime%>01:23:45
event ok
else
event nok
endif
endon

```
Log:   
```
>>Send to /dev/ttyUSB1: "event oldtime"<<
>event oldtime
EVENT: oldtime
to check: 23:03:39>12:34
12:34 String -> Float 12.00
23:03:39 String -> Float 23.00
compare: > match
ACT  : event ok
EVENT: ok

Ok

>>Send to /dev/ttyUSB1: "event newtime"<<
>event newtime
EVENT: newtime
to check: 23:03:52>01:23:45
01:23:45 String -> Float 1.00
23:03:52 String -> Float 23.00
compare: > match
ACT  : event ok
EVENT: ok

Ok
```

more info on #592

TD-er, 2018-02-26:
See this code:
https://github.com/letscontrolit/ESPEasy/blob/076ce0127b9ce116e71931d77e0ffbeaadafa265/src/Misc.ino#L2217-L2245

Instead of just calling the `toFloat()` function, first check if it is a float (`isFloat()` ) and if not, convert it to seconds and use that as `ruleValue`.

uzi18, 2018-02-27:
```
>>Send to /dev/ttyUSB1: "event oldtime"<<
>event oldtime
EVENT: oldtime
to check: 01:42:18>12:34
 String -> Float 6138.00
 String -> Float 45240.00
compare: > not match
ACT  : event nok
EVENT: nok

Ok

>>Send to /dev/ttyUSB1: "event newtime"<<
>event newtime
EVENT: newtime
to check: 01:42:27>01:23:45
 String -> Float 6147.00
 String -> Float 5025.00
compare: > match
ACT  : event ok
EVENT: ok

Ok

```

uzi18, 2018-02-27:
#964 is ready.

commit f0fc20de (2018-02-27): fix #964, add timeStringToSeconds and use it to compare time conditions (#965)

* - pcfgpio,port,2 sets pin as input
- fix for #222,#702 - we can mix inputs and outputs now

* - extend conditions with AND/OR, now it is possible to simplify rules

test rules:
on test do
  if [test#a]=0 or [test#b]=0 or [test#c]=0
    event ok
  else
    event nok
  endif
endon

on test2 do
  if [test#a]=1 and [test#b]=1 and [test#c]=1
    event ok
  else
    event nok
  endif
endon

on test3 do
  if [test#a]=1 and [test#b]=1 or [test#c]=0
    event ok
  else
    event nok
  endif
endon

on test4 do
  if [test#a [... cut]
commit e86667e3 (2018-03-01): fix #964, add timeStringToSeconds and use it to compare time conditions (#965)

* - pcfgpio,port,2 sets pin as input
- fix for #222,#702 - we can mix inputs and outputs now

* - extend conditions with AND/OR, now it is possible to simplify rules

test rules:
on test do
  if [test#a]=0 or [test#b]=0 or [test#c]=0
    event ok
  else
    event nok
  endif
endon

on test2 do
  if [test#a]=1 and [test#b]=1 and [test#c]=1
    event ok
  else
    event nok
  endif
endon

on test3 do
  if [test#a]=1 and [test#b]=1 or [test#c]=0
    event ok
  else
    event nok
  endif
endon

on test4 do
  if [test#a [... cut]
closed 2018-02-27 by psy0rz