letscontrolit/ESPEasy #967: MQTT: publish topic override for each task

M0ebiu5, 2018-02-27:
Each task should have a field to override the default mqtt topic.

TD-er, 2018-02-27:
Such a change has quite some impact on how the task settings are being stored.
And how long should a topic string be?
Should it append to the given topic in the used controller?

Grovkillen, 2018-02-27:
I use rules for that. I have the "send to controller" unchecked and publish from rules.

M0ebiu5, 2018-02-27:
The new topic should override and not append the default topic.
For me, 40 chars would be enough.

Nice workaround with the rules, but not obvious for a normal user :)
Also, it uses more resources.

M0ebiu5, 2018-03-13:
Additionally, there should also be a possibility to set the "retain" flag individually per task topic (checkbox).

Grovkillen, 2018-03-13:
Maybe extend the publish command with QoS and Retain?

TD-er, 2018-03-13:
Maybe create a new keyword for it "Publish_ret" ? That makes it more distinguishable.

Grovkillen, 2018-03-13:
If it's better code wise then I agree :+1:

M0ebiu5, 2018-03-13:
now it's implemented like this:
`
String topic = event.substring(0, index);String value = event.substring(index + 1);
`

TD-er, 2018-03-13:
And then keep the global retain flag setting to enable/disable its use globally? (on the ESP module)

M0ebiu5, 2018-03-13:
good question - one way would be to use the global flag, if none is explicitly set - otherwise use the parameter value from the publish cmd. This way it's possible to override the global flag in both ways.

raceboy, 2018-06-01:
i miss retain flag in rules too. is it possible to easy change in code or it's complicated ?

PR #4984 (2024-02-21): [MQTT] Add PublishR command: publish with will-retain
Resolves #1718 
Resolves #967 

Features:
- Add command `PublishR,<topic>,<value>` to send a value to a topic on the first enabled MQTT Controller
  _with the Will Retain flag enabled_
- Updated documentation
closed 2024-02-21 by TD-er