letscontrolit/ESPEasy #968: Degrees symbol not parsed for OLED display when triggered by MQTT command

ax42, 2018-02-27:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Send MQTT command `OLED,6,1,{D}C` to controller.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
I should get `<degree symbol>C` in the bottom left of the display.  

### Actual behavior
I get `{D}C`.

### System configuration
Hardware:

WeMos D1 and Wemos OLED shield (which all works fine otherwise).

Software or git version: 

Build | 20100 - Mega (core 2_3_0)
GIT version | mega-20180224
Plugins | 48 [Normal]

TD-er, 2018-02-27:
Did you also try this one:
`&deg;`
Or any of the other special characters in [this list](https://www.letscontrolit.com/wiki/index.php/OLEDDisplay#Special_characters) ?

And what display plugin do you use?
The OLED (without "framed") uses a different font, so perhaps not all symbols are included.
The Framed one doesn't have commands to display text, if my memory serves me right.

ax42, 2018-02-27:
`&deg;` also fails.  The `{D}` works fine as configured from within the web interface (as does `&deg;`), just not via MQTT.

Just tried the `{D}` command via HTTP as well (`http://<ip>/control,6,1,88{D}`), same problem.

I'm using the `Display - OLED SSD1306` module.

TD-er, 2018-02-27:
If the `&deg;` notation also doesn't work via HTTP, then it is just a matter of calling the replacement functions.
If that works via HTTP, then it is a problem with the allowed characters in the protocol parsers.

The "{" character may not be permitted in a HTTP get request.

ax42, 2018-02-27:
Thanks.  Sending `%7BD%7D` (ie encoding the brackets in HTTP), also just gives `{D}` in the display.

Sending the unicode degree symbol `°` in the HTTP request gives two filled squares in the display.

Not quite sure what you mean with "it is just a matter of calling the replacement functions."
is this something I should be trying to do?

TD-er, 2018-02-27:
Nope, a mental note for me, to fix it. :)

TD-er, 2018-02-27:
Fixed and even added more possibilities in PR #974
It should even be possible to use task values in commands.

TD-er, 2018-03-01:
{D}
See the "?" Symbol next to the plugin name.
That links to the wiki page. Scroll to the bottom of that page for a list of supported symbols

mattlward, 2018-03-01:
I do not, that links to the OLED page, are the FRAMEDOLED values the same?

Grovkillen, 2018-03-01:
The values are the same. We haven't had the time to write a wiki for framed.

mattlward, 2018-03-01:
Is that something I could help with?

Grovkillen, 2018-03-01:
Sure! Either use me(or Shardan) and send the text or create a wiki account.

TD-er, 2018-03-23:
I guess this one can be closed.
If not , please re-open.

closed 2018-03-23 by TD-er