letscontrolit/ESPEasy #969: MQTT: add ability to subscribe to more than one topic

M0ebiu5, 2018-02-27:
One espeasy device should be able to subscribe to multiple topics.
This would allow easy grouping and synchronisation (eg switching multiple lights).

Grovkillen, 2018-02-27:
a more generic way is discussed here: #378

M0ebiu5, 2018-02-27:
ic, but is this way not easier to implement and understand?
If the underlying mqtt lib already allows multiple subscriptions, it would be sufficient to just enter in the topic field a comma separated list of the topics...

Grovkillen, 2018-02-27:
If it's easier to implement I don't know but imagine explaining the "open" approach you suggest. Many would not understand that either but a highly standardized ESP Easy approach with groups etc. would at least be defined.

TD-er, 2018-02-27:
Those settings must be stored somewhere and the current storage layout of settings does not allow dynamic changes in data to be stored.

Also, what part should do the subscribes? The controller, or the MQTT importer?
And how to interact with the different subscribes?

M0ebiu5, 2018-02-27:
@TD-er sorry, but i dont understand where there is a need to store dynamic changes and also, your other questions. 

Let's take this example:
The espeasy device subscribes to the topics "tvlight","light/livingroom" and "light/firstfloor"
Now if i want to turn off/on all lights in the first floor, i can submit the command to "light/firstfloor"- to only address the device, i would use "tvlight" and so on.

Nothing else changes - the 3 subscriptions just allow to send one command to a different group of devices.

TD-er, 2018-02-27:
But MQTT supports such kind of behavior by adding a wildcard between topic levels.

It is best described [here](https://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices)

What I mean is that these extra topics should be stored somewhere in the settings and that's currently very hard to change. The way settings are stored is not that flexible.

M0ebiu5, 2018-02-27:
How can i use wildcards to publish a command? How this works with the example i gave?

The current topic is already stored somewhere - whats the difference to store a comma separated string with multiple topics on the same place?

Grovkillen, 2018-02-27:
Please understand that the architecture of the settings are 1-to-1, bit by
bit.

It's a very limited device we're cramming the *hit out of.

TD-er, 2018-02-27:
@M0ebiu5 The idea is that you can listen to more than one topic using wildcards.
You subscribe to a topic using wildcards. You publish only to one.

And storing a comma separated list is taking extra space. That could be working if it fits in the dedicated space to store the topic, only I have to change the interpreter of  that setting to create several topics from a line containing a comma separated list.
The thing is, the allowed max topic length is fixed (32 or 64 Bytes probably) and you cannot exceed that amount without changing a lot.

M0ebiu5, 2018-02-27:
It's difficult to have a discussion, if many questions stay unanswered...
Anyway, thanks for your time!

TD-er, 2018-02-27:
Please try to summarize your questions with some usecases, perhaps I don't understand your questions.
And please read the hiveMQ link I gave.
I think what is described there is what you want to do.
When ESPeasy should decide to give commands, you should do it from a rule and publish that to a specific topic (or set of topics)
When ESPeasy should listen to multiple topics (subscribe), it is probably best to have the topic levels aligned to be able to add some wildcard to a specific level. (e.g. "all" at the same level as a specific room) 
It may be possible to add some comma separated list, but I am not sure that will conflict with current MQTT standards. Also you cannot exceed the length allowed to store in the input field (e.g. 64 bytes) since then you will overwrite settings of other plugins or controllers.

fmuntean, 2018-07-21:
I vote for this one too.
would be nice to be able to specify another topic to subscribe to.
One use case would be a broadcast channel where I can send info to a group of devices.
Currently that can only be handled outside sending the same MQTT message to each device topic separately.
So in the end the MQTT controller will be able to subscribe to the device topic and a second group topic.

with this feature will be able to group the devices as needed. 
Another option would be to add some kind of command that we can call on the MQTT#connect event to manually subscribe to other topics.

closed 2023-09-04 by TD-er