letscontrolit/ESPEasy #970: MQTT import not starting properly.

mattlward, 2018-02-27:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

TD-er, 2018-02-27:
How many "MQTT import tasks" do you have?
Only one can currently be active.

mattlward, 2018-02-27:
That is interesting.. I have 2 running on a couple of units and they work fine.  But this problem also spans single instance units.

TD-er, 2018-02-27:
The thing is, I added preliminary support for allowing to use >1 MQTT.
But currently not everywhere in the code is the "active" MQTT controller known.
So it might be possible a second instance is using the settings of the first enabled MQTT controller.

If they connect to the same broker, then it is not really an issue, but they may also share the same topics and the "connect" status.
So if one of them looses connection, it is very well possible only the first enabled one will reconnect and the others will not.

mattlward, 2018-02-27:
Ok, yes they use the same broker.

TD-er, 2019-11-04:
This is no longer an issue, so I will close this one.

closed 2019-11-04 by TD-er