letscontrolit/ESPEasy #971: P141 LedStrip Plugin compiles with Errors

micropet, 2018-02-27:
When compiling with the Arduino IDE 1.8.5, the following errors appear:
The plugin is from ESPEasyPluginPlayground.
Since there is no issue tracker, I'll report it here.

_P141_LedStrip:719: error: 'CRGB' does not name a type

 void Fp141_OutputRGB( const CRGB& rgb){

                             ^

_P141_LedStrip:719: error: ISO C++ forbids declaration of 'rgb' with no type [-fpermissive]

 void Fp141_OutputRGB( const CRGB& rgb){

                                   ^

_P141_LedStrip:741: error: variable or field 'Fp141_OutputHSV' declared void

 void Fp141_OutputHSV(CHSV hsv){

                      ^

_P141_LedStrip:741: error: 'CHSV' was not declared in this scope

_P141_LedStrip:760: error: 'CHSV' does not name a type

 CHSV Fp141_RgbToHSV(CRGB rgb){

 ^

_P141_LedStrip:765: error: variable or field 'Fp141_SetCurrentColor' declared void

 void Fp141_SetCurrentColor(CHSV hsv){

                            ^

_P141_LedStrip:765: error: 'CHSV' was not declared in this scope

_P141_LedStrip:776: error: 'CRGB' does not name a type

 CRGB Fp141_CharToRgb(const char * rgb) {

 ^

_P141_LedStrip:786: error: 'CHSV' does not name a type

 CHSV Fp141_CharToHsv(const char * rgb) {

 ^

_P141_LedStrip:796: error: 'CRGB' does not name a type

 CRGB Fp141_LongToRgb(unsigned long rgb){

 ^

_P141_LedStrip:805: error: 'CHSV' does not name a type

 CHSV Fp141_LongToHsv(unsigned long hsv){

 ^

In file included from D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino:72:0:

D:\arduino-1.8.5-ESPEasy\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.008

 #    pragma message "FastLED version 3.001.008"

                     ^

In file included from D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P064_APDS9960.ino:31:0:

D:\arduino-1.8.5-ESPEasy\libraries\SparkFun_APDS-9960_Sensor_Arduino_Library\src/SparkFun_APDS9960.h:109:33: error: expected '>' before numeric constant

 #define WAIT                    3

                                 ^

D:\arduino-1.8.5-ESPEasy\libraries\FastLED/fastled_delay.h:13:14: note: in expansion of macro 'WAIT'

 template<int WAIT> class CMinWait {

              ^

In file included from D:\arduino-1.8.5-ESPEasy\libraries\FastLED/FastLED.h:68:0,

                 from D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino:72:

D:\arduino-1.8.5-ESPEasy\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'boolean Plugin_141(byte, EventStruct*, String&)':

_P141_LedStrip:454: error: 'Fp141_RgbToHSV' was not declared in this scope

     Fp141_SetCurrentColor( Fp141_RgbToHSV(CRGB(event->Par1, event->Par2, event->Par3)));

                                                                                      ^

_P141_LedStrip:454: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor( Fp141_RgbToHSV(CRGB(event->Par1, event->Par2, event->Par3)));

                                                                                       ^

_P141_LedStrip:459: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(CHSV (event->Par1, event->Par2, event->Par3));

                                                                       ^

_P141_LedStrip:465: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(v_p141_cur_color);

                                           ^

_P141_LedStrip:471: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(v_p141_cur_color);

                                           ^

_P141_LedStrip:477: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(v_p141_cur_color);

                                           ^

_P141_LedStrip:487: error: 'Fp141_CharToRgb' was not declared in this scope

     v_p141_cur_color = Fp141_RgbToHSV(Fp141_CharToRgb( color.c_str() ));

                                                                      ^

_P141_LedStrip:487: error: 'Fp141_RgbToHSV' was not declared in this scope

     v_p141_cur_color = Fp141_RgbToHSV(Fp141_CharToRgb( color.c_str() ));

                                                                       ^

_P141_LedStrip:488: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(v_p141_cur_color);

                                           ^

_P141_LedStrip:495: error: 'Fp141_CharToHsv' was not declared in this scope

     v_p141_cur_color = Fp141_CharToHsv( color.c_str() );

                                                       ^

_P141_LedStrip:496: error: 'Fp141_SetCurrentColor' was not declared in this scope

     Fp141_SetCurrentColor(v_p141_cur_color);

                                           ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_AnimFlash(boolean, byte)':

_P141_LedStrip:606: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV(CHSV {0,0,0});

                               ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_AnimStrobe(boolean, byte)':

_P141_LedStrip:625: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV(CHSV {0,0,0});

                               ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_AnimFade(boolean, byte)':

_P141_LedStrip:637: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV( CHSV(v_p141_cur_color.h, v_p141_cur_color.s, dim8_lin(v_p141_cur_anim_step)) );

                                                                                                 ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_AnimSmooth(boolean, byte)':

_P141_LedStrip:670: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV(v_p141_cur_anim_color);

                                        ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_AnimParty(boolean, byte)':

_P141_LedStrip:691: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV(v_p141_cur_anim_color);

                                        ^

_P141_LedStrip:696: error: 'Fp141_OutputHSV' was not declared in this scope

   Fp141_OutputHSV(CHSV {0,0,0});

                               ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_CommandOn()':

_P141_LedStrip:706: error: 'Fp141_SetCurrentColor' was not declared in this scope

  Fp141_SetCurrentColor(v_p141_cur_color);

                                        ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'void Fp141_CommandOff()':

_P141_LedStrip:714: error: 'Fp141_SetCurrentColor' was not declared in this scope

  Fp141_SetCurrentColor(v_p141_cur_color);

                                        ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'CRGB Fp141_CharToRgb(const char*)':

_P141_LedStrip:782: error: 'Fp141_LongToRgb' was not declared in this scope

     return Fp141_LongToRgb( strtoul(p, NULL, 16) );

                                                  ^

D:\arduino-1.8.3\meine\ESPEasy-mega-02-25\ESPEasy\_P141_LedStrip.ino: In function 'CHSV Fp141_CharToHsv(const char*)':

_P141_LedStrip:792: error: 'Fp141_LongToHsv' was not declared in this scope

     return Fp141_LongToHsv( strtoul(p, NULL, 16) );

                                                  ^

Mehrere Bibliotheken wurden für "ESPeasySoftwareSerial.h" gefunden
 Benutzt: D:\arduino-1.8.5-ESPEasy\libraries\ESPeasySoftwareSerial
 Nicht benutzt: D:\arduino-1.8.5-ESPEasy\libraries\SoftwareSerial
exit status 1
'CRGB' does not name a type

TD-er, 2018-02-27:
P141 is from the playground, not this project.

micropet, 2018-02-27:
OK, sorry.

closed 2018-02-27 by micropet