letscontrolit/ESPEasy #972: ESP32 DS18b20 sensor problem

redskinhu, 2018-02-27:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Connect DS18b20 temperature sensor to any suitable GPIO.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
reboot/reset not helps

### Expected behavior
Working sensor.

### Actual behavior
On the ESPEasy32 DS18b20 webpage the serial nr of the sensor appearing after the first submit. But if i choose the sn of the DS18b20 and submit the page again the serial nr is disappearing from the field and never comes back.

### System configuration

Hardware:
ESP32

Software or git version:
Build: 20000 - Mega32

TD-er, 2018-02-27:
Huh, do we have a build for the ESP32?

Grovkillen, 2018-02-27:
No, we are not ready to take care of ESP32 users yet. Totally committed to ESP82xx and that's plenty of work. :wink:

TD-er, 2018-02-27:
But I am really curious what version reports "Build: 20000 - Mega32"
Is it the Python implementation of Lisa?

Budman1758, 2018-02-27:
@TD-er  See this link
[https://www.letscontrolit.com/wiki/index.php/ESPEasy32](url)

TD-er, 2018-02-27:
Is that @mvdbro 's project to get it running on ESP32?

Well as the topic starter probably already understood, I cannot help here :(

Grovkillen, 2018-02-27:
Yes it's @mvdbro who are the main developer for that but I don't know what he's status on support is..

Budman1758, 2018-02-27:
Don't think there is any support to speak of yet.  Seems like the whole thing needs its own repository and github setup.

redskinhu, 2018-02-27:
Yep. Exactly. I tried the 20100 too, but that is not stable. Restarts 1-4 hourly. I tried to compile one, but no luck. :(

(This is just a feedback, I don't expect any immediate action.)

Anyway I'm a happy ESPEasy32 dev/user. :-)

Thx

mvdbro, 2018-03-01:
As one can see from the Wiki link, the ESP32 route is highly experimental.
Only some major features from the ESP Easy Framework have been tested.

Currently all active developers are working on ESP8266 only. Getting the code compiled is a small challenge after every commit anyway because dev code isn't verified against ESP32 by travis.

Future of ESP Easy on ESP32 isn't settled yet. It may just be easier to use uPyEasy on the ESP32.

Budman1758, 2018-03-03:
By the way, I can confirm this bug as described.  Does the exact same thing on the unit I am testing.  Framed OLED is working though.  :)

mvdbro, 2018-03-03:
Fixed

closed 2018-03-03 by mvdbro

redskinhu, 2018-05-09:
Hello

At the end I realized how can I compile (custom) ESP32 fw.

Compiled Release mega-20180506 to ESP32.

The bug is still exisit. It changed a little: I couldn't see the sn of the DS18b20 and if I want to change the GPIO pin to another pin (which is not connected to DS18b20) the ESP32 restarts.

I saw this issue is fixed and closed but it looks not or it is a new issue.

Could you check it?

Thanks