letscontrolit/ESPEasy #977: MQTT Auto-discovery feature suggestion !

vincegre, 2018-02-28:
Following a thread opened in forum https://www.letscontrolit.com/forum/viewtopic.php?p=25237#p25036, I was suggested to repost it here ;)

More and more automation systems are able to use the auto-discovery feature with MQTT broker (so basically you don't need to declare one by one all your sensors, switches,... in your automation system, they are automatically created with correct type first time your ESP is going to connect to the MQTT broker of your system).

So here is the cooking book for it ;)

- First

 you need to setup the root prefix for autodiscovery (should be a field to fill). For Home Assistant it's homeassistant basically. That field should be combined with a button to "activate" the discovery and a unique node ID (user fileable).

- Second

 Once discovery button has been pushed and device restarted, it'll send automatically on MQTT broker setup in ESPEasy a configuration topic for each hardware device configured in ESPEasy (Home Assistant for example supports in autodiscovery binary sensors, covers, fans,lights, sensors and switches !!). The configuration topic should be formatted this way:
[code]<discovery_prefix>/<component>/[<node_id>/]<object_id>/<>[/code]
"component" being one of the type previously listed
"node_id" being an optional field to supply node providing the topic
"object_id" being the unique node ID filled by user in ESPEasy parameters (if not filled can be by default the last digits of the MAC address of the ESP for example)
"<>" being the config or state to define current action. 

- Third

 The discovery button will be putted back in OFF state in ESPEasy interface.

For more details with Home assistant (basically same process with all MQTT clients that support auto-discovery): https://home-assistant.io/docs/mqtt/discovery/

If you need a tester I'm your man :)

vincegre, 2018-03-07:
Any chance to get it implemented soon in the dev or tests builds ? Thanks

vincegre, 2018-03-19:
up on this ;)

TD-er, 2018-03-19:
I guess this won't be done before the 2.0 release.
We're working as hard as we can on fixing all kinds of issues and also some usability related ideas.
But not sure who would do this.
I am not running HomeAssistant and also MQTT is not my kind of expertise.

vincegre, 2018-03-22:
Sad to hear the news but I expected that :( hope it can be avalaible soon in beta build still :D

Naesstrom, 2018-04-03:
I actually came to make the same suggestion here, could be doable right now by just adding the correct syntax in the topic I guess but would love it as a feature. Same time add home-assistant mqtt as an option instead of using some of the other ones.

Misiu, 2019-10-24:
I'm using Home Assistant for some time now and I found out that it [supports MQTT discovery](https://www.home-assistant.io/docs/mqtt/discovery/).
Current Home Assistant (openHAB) controller works fine, but it requires us to configure everything manually, for example, a simple temperature sensor needs this:

```
  - platform: mqtt
    state_topic: "Bedroom_TandH/Bedroom/Temperature"
    name: "Bedroom"
    unit_of_measurement: "°C"
    expire_after: 60
    #force_update: true
    availability_topic: "Bedroom_TandH/status"
    payload_available: "Online"
    payload_not_available: "Offline"
```

It would be awesome if ESPEasy would support discovery, this was all the sensors could be automatically detected.

Tasmota, ESPHome, ESPurna, and others support this. I'm not listing them here to compare ESPEasy to them, but I'm sure the implementation would be very similar and maybe bits of code can be taken from there 😉

TD-er, 2019-10-24:
Some of this auto  discovery is already present, but since I have not made it and not used it, I cannot tell you about the current state of it.
It is included in controller 14 if my memory serves me well and it also has a plugin to go along with it. (P086???)

Misiu, 2019-10-24:
@TD-er at least I have a starting point :)
I have two Home Assistant instances and couple of Wemos D1 with ESP Easy on it so I'll give this a try.
I never used discovery in HA, but I think that now is the time 🙂

CyberStar1, 2019-11-19:
Any updates on this matter? :) Would love to see this feature in the future of ESPEasy.

uzi18, 2020-02-25:
@Misiu do you have any info where to find how to configure any sensor manually in HA?

Misiu, 2020-02-25:
@uzi18 if you are using MQTT then this is quite easy.
https://www.home-assistant.io/integrations/sensor.mqtt/ is a good starting point.
I used MQTT.fx to get all the topics and configured them in Home Assistant.
I have 4 DS18B20 sensors working about 5 months right now, so this is quite stable.

uzi18, 2020-02-25:
just installes HA on my RPi3 and looking for clues, thx will read it

Misiu, 2020-02-25:
@uzi18 no problem :)

Mav63, 2020-12-31:
Any updates on this until now?
It would be very neat to have this feature up and running soon since other home automation SW using this feature heavily (i.e. openHab 3).
Thanks for your support here... 
BTW: If you need testing, I am open (using a couple of NodeMCUs with various sensors, like DHT22 , DS18B20, analog LDRs etc.)!

tonhuisman, 2020-12-31:
As mentioned before, the C014 `Homie MQTT` controller already sends all device data (System info, GPIO's and enabled Devices) on connection (initial and reconnect) to the server. As I don't know much about MQTT, I can only report that it sends a lot of $-prefixed topics to the server, containing the related settings. As I only have Domoticz installed (and no intention to setup Homey or OpenHAB any time soon), I see no way to validate this, ATM. Can't tell if this format is outdated or anything, if so then maybe we could invite the original developer to update 🤔

KnatteAnka, 2021-01-27:
In the mean time or as an alternative I am working on an Node-red page that checks incoming data and creates sensors in HA if needed.
but what i miss is an easy way to set a variable in EE with like /set to the variable or such and not needing to manually input which gpio or use events on EE.

Currently beta version is working for sensors without Anything on an EE more then setting up the controller and enabling it to send the device.
and switches or relays needs a event to set the gpio+variable.
light and

tonhuisman, 2021-01-27:
> what i miss is an easy way to set a variable in EE with like /set to the variable or such and not needing to manually input which gpio or use events on EE.

You mean like this PR #3098 tries to solve?

KnatteAnka, 2021-01-27:
or just 
EE/FooBar/ADevice/AValue/Set
meaning ~/Set = a taskvalueset command is sent with the payload to the value before it

KnatteAnka, 2021-01-27:
but what i can and will do for the time being is using an dummy device to store output value
and an MQTT import for reading inputs and and event for toggle of the switch

KnatteAnka, 2021-01-29:
if anyone want to test the current node red node i have uploaded it to github:
https://github.com/KnatteAnka/NodeRED_ESPEASY_AutoDiscovery

poudenes, 2021-11-20:
Is this request still on scope? I would like to see MQTT discovery as well into ESPeasy.
Now have a workaround to let ESPeasy work with HA

KnatteAnka, 2021-11-21:
i do not see a need to have full AutoDiscovery function fully in EE, but an easy way to set values and not need to use rules would be great. 
(At the moment im using rules to set values with my node red AutoDiscovery flow
like 
MQTT out for a value is: EE/Unit1/Task1/Var1
To set that send value to: EE/Unit1/Task1/Var1/Set (Not working atm)

gargamelonly, 2022-11-19:
Any updates on this until now?

closed 2025-07-02 by TD-er