letscontrolit/ESPEasy #978: Controller Subscribe and Controller Publish always get replaced

killler718, 2018-02-28:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
System variables always get replaced with empty string

Does the problem persist after powering off and on? (just resetting isnt enough sometimes)
yes

### Expected behavior
Tell us what should happen?
Entering /%sysname%/# to subscribe should remain

### Actual behavior
Entering /%sysname%/# to subscribe gets replaced with /
this hold for publish as well 
/%sysname%/%tskname%/%valname% gets replaced with / and this is where it is send to controller

### System configuration

Hardware:
wemos D1 mini
Software or git version:
git mega

TD-er, 2018-02-28:
What version of ESPeasy did you use? (date of build or the date you downloaded the sources)

killler718, 2018-02-28:
Git 4e2a5d3b3d3ce6eabb091ec8cdd1b2d54b2dc440 HEAD -> mega, tag: mega-20180228, origin/mega
just built. 
v2.0 same thing is happening i just can't figure out how to trace it. Suspect its an issue with StringConverter but can't find where the parseSystemVariables() is called

TD-er, 2018-02-28:
> Suspect its an issue with StringConverter but can't find where the parseSystemVariables() is called

Probably nowhere ;)
I will look into it this evening.

killler718, 2018-02-28:
OK. Formula %value% same behaviour just noticed

killler718, 2018-02-28:
Managed to run some kind of debug:
getting this :
args: index=1&protocol=5&usedns=0&controllerip=10.1.1.9&controllerport=1883&controlleruser=sensor&controllerpassword=*****&controllersubscribe=/ATEST/%sysname%/#&controllerpublish=/ATEST/%sysname%/%tskname%/%valname%&controllerenabled=on
args count: 10
pos 0=@ 5 &@ 7
arg 0 key: index value: 1
pos 8=@ 16 &@ 18
arg 1 key: protocol value: 5
pos 19=@ 25 &@ 27
arg 2 key: usedns value: 0
pos 28=@ 40 &@ 53
arg 3 key: controllerip value: 10.1.11.9
pos 54=@ 68 &@ 73
arg 4 key: controllerport value: 1883
pos 74=@ 88 &@ 95
arg 5 key: controlleruser value: sensor
pos 96=@ 114 &@ 120
arg 6 key: controllerpassword value: *****
pos 121=@ 140 &@ 159
arg 7 key: controllersubscribe value: /ATEST/
pos 160=@ 177 &@ 214
arg 8 key: controllerpublish value: /ATEST/  sname  tskname  valname%
pos 215=@ 232 &@ -1
arg 9 key: controllerenabled value: on
args count: 10

looks like the args are not properly parsed. Still looking where is this actually happening.

killler718, 2018-02-28:
OK forget this. issue is with the ESP8266 Arduino core i was using the wemos one which apparently is way outdated. Switching to https://github.com/esp8266/Arduino resolves all issues with WebServer.

closed 2018-02-28 by killler718