letscontrolit/ESPEasy #979: Json length reported as 0

Grovkillen, 2018-02-28:
Node Red is having problem with latest json get. The content length (in header) is reported as 0 instead of the actual length of response body.

Grovkillen, 2018-02-28:
It is reported to work with mega-20180118

TD-er, 2018-03-01:
It looks like the new webserver is always returning `Content-Length:0`
Perhaps it has something to do with the way the chunked transfers are being dealt with?

Maybe @s0170071 can give some more info on this?

Isaksson, 2018-03-02:
It could be better when using chunked encoding to not set the Content-Length header at all instead of setting it to 0.

TD-er, 2018-03-02:
Ik looks like the Content-Length header is also set elsewhere.
Last night, I disabled it in the Webserver part, but it was still there.
It is set twice, the same for the Content-Type.

s0170071, 2018-03-02:
Hold on. I'll fix it

TD-er, 2018-03-02:
I did some basic layouting of your part last night
It was totally unreadable due to mixed tabs and spaces.

So please fetch that file first before editing.

s0170071, 2018-03-02:
The json file is delivered as chunked, so there is no content length. Unless this is not allowed for json in general, this should work. Its using the same routine like any other html file. 

In fact it does work with firefox using mega 2.4.0 and 2.3.0 core. At least I an't reproduce any error. 

@Grovkillen, what's your setup (IDE, browser, OS, core) ? 
What you can try is replacing the handle_json() function with the version from january and test if that works for you. Its a stand-alone routine which handles the header and send by itself.
It that still does not work I would suggest to erase the config and  re-configure from scratch. This fixed some problems for me already.

TD-er, 2018-03-02:
You can see quite easily what's going on by using the inspect tool in the browser.

As a comparison, you can also have a look at the Json output of a Django project of mine, located [here](http://ovbureau.kxa.nl/api/line/a036)
Then right-click => inspect => Network-tab => refresh in the browser and click on the item retrieved in the inspect view.

[image]

You can do the same with the /json url on the ESPeasy board.
Then you'll see there are 2x Content-Type.
In the response from my Django example, you can see no Content-Length is reported.
So I guess if you remove it, it will be accepted again.

Isaksson, 2018-03-02:
Yes, that is true when you set the Transfer-Encoding header to chunked then the header Content-Length should not be set at all.

s0170071, 2018-03-02:
Reverting to the original, non chunking json routine is an option. Just tested, it would work.

Isaksson, 2018-03-02:
Non chunking would be good enough, I do not think that chunking will do much of a differens for that function.

TD-er, 2018-03-02:
Chunking is good for big streams of data. Currently the Json output is not that big, but may become bigger in the future when all settings may be dumped via Json.
I think the httpserver in the Arduino core libraries is inserting this value.
You could set the size to be unknown, perhaps that will refrain that library to send the length header?

s0170071, 2018-03-02:
fixed in PR #988

TD-er, 2018-03-02:
I think we still should try to use your approach to do chunked transfers.
But that's for later.
For now, it's a quick fix.

TD-er, 2018-03-02:
Fixed in PR#988.

djcysmic, 2018-03-03:
I still get no JSON answer in FHEM. Every 4 times I send a command I get the last 4 JSON answers on HTML, but with no JSON content...
[image]

TD-er, 2018-03-03:
Hmm, did not know there was also Json output when sending commands.
I only looked into "/json" url.

Maybe we should make a Json formatter of some kind?

djcysmic, 2018-03-03:
I have two plugins working with a JSON answer because I need non-integer answers:

```
// ---------------------------------------------------------------------------------
// ------------------------------ JsonResponse -------------------------------------
// ---------------------------------------------------------------------------------
void Plugin_123_SendStatus(byte eventSource)
{
  String log = String(F("Lights: Set ")) + Plugin_123_pins[0].FadingTargetLevel
             + String(F("/")) + Plugin_123_pins[1].FadingTargetLevel + String(F("/")) + Plugin_123_pins[2].FadingTargetLevel
             + String(F("/")) + Plugin_123_pins[3].FadingTargetLevel + String(F("/")) + Plugin_123_pins[4].FadingTargetLevel;
  addLog(LOG_LEVEL_INFO, log);

  String onOff = (Plugin_123_lightParam.state) ? "on" : "off";
  String cm = (Plugin_123_lightParam.colorMode == 2) ? "ct" : "rgb";

  String json;
  printToWebJSON = true;
  json += F("{\n");
  json += F("\"plugin\": \"123");
  json += F("\",\n\"onOff\": \"");
  json += onOff;
  json += F("\",\n\"rgb\": \"");
  json += Plugin_123_lightParam.rgbStr;
  json += F("\",\n\"pct\": \"");
  json += Plugin_123_lightParam.pct;
  json += F("\",\n\"ct\": \"");
  json += Plugin_123_lightParam.ct;
  json += F("\",\n\"colormode\": \"");
  json += cm;
  json += F("\"\n}\n");
  SendStatus(eventSource, json); // send http response to controller (JSON fmormat)
}
```
But with the last ESP Easy versions I miss these answers...

How could we realize this JSON formatter? Do you mean a general function in ESP Easy which could be used by every plugin?

TD-er, 2018-03-03:
@djcysmic That's a plugin of the playground and not part of the normal ESPeasy branch.

And indeed, I was thinking about a general function/class to do the JSON formatting and combine it with the great streaming code for the Webserver written by @s0170071 .
My plan is to do a lot more with JSON, so a generic way of generating it would be more than welcome.

djcysmic, 2018-03-03:
> @djcysmic That's a plugin of the playground and not part of the normal ESPeasy branch.

That's correct. It worked very well, but I had the problem with issue [#622](https://github.com/letscontrolit/ESPEasy/pull/622). So I updated ESP Easy and after that update I ran into this issue.

TD-er, 2018-03-03:
OK, that's caused by the introduction of the "2k webserver".
In short, that's a class to output chunked data in short bursts. That's much more memory friendly.
But apparently the main HTTPserver classes in the core library also add headers, which are incompatible with this approach.
For browsers this is not an issue, but other applications dealing with JSON are more strongly checking these headers.

For the /json url this is fixed by creating one big String like before and not using the streaming webserver.

djcysmic, 2018-03-03:
Thank you for this information - I got a solution for me:

I changed in WebServer.ino line 3407:

`WebServer.send(200, "application/json");`
to
`WebServer.send(200, "application/json", printWebString);`

TD-er, 2018-03-03:
Not sure if that will work, since the string is still written to the streaming webserver and that will be flushed also at some moment.

djcysmic, 2018-03-03:
For the moment it is running good - at least I got an answer with the state of my devices in FHEM.

Grovkillen, 2018-03-08:
Tested and working

closed 2018-03-08 by Grovkillen

TD-er, 2018-03-08:
And I had just plans to change it back to using the streamed class, since I now add the proper length at the end.

Grovkillen, 2018-03-08:
Aha, might try that as well.