letscontrolit/ESPEasy #986: Add Support for CDM7160 / Figaro sensor

Egyras, 2018-03-02:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?
Add Support for CDM7160 / Figaro sensor

### Actual behavior
Tell us what happens instead?
Add Support for CDM7160 / Figaro sensor
### System configuration

Hardware:
CDM7160 / Figaro sensor

Software or git version:
I found some source code for faster implementation:
https://github.com/vlast3k/vESPrino-fir ... rc/sensors
 CDM7160Sensor.cpp 
 CDM7160Sensor.hpp
https://github.com/junichikatsu/CDM7160

Egyras, 2018-03-02:
I'am waiting for actual sensor to test. But according forum thread it doenst have support.

TD-er, 2018-03-02:
That's right, it isn't. Was just joking.  I'll have to order one myself too, in order to support it.

Egyras, 2018-03-02:
I have limited coding expierience, but try to do my best when get sensor and some free time.

TD-er, 2018-03-07:
See [this forum post](https://www.letscontrolit.com/forum/viewtopic.php?p=25639#p25639):

> Unread post  by Vaikis » 07 Mar 2018, 11:18
> 
> Got sensor and some 30min time to play, tips for connection for other guys: 
> 1. Used pull-up resistors 1K for both SCL (CDM7160 PIN 10) and SDA (CDM7160 PIN9)
> 2. CDM7160 PIN1 - VDD +5V, CDM7160 PIN2 GND
> 3. Dont forget to enable I2C as default mode is UART, CDM7160 MSEL (PIN6) connect to GND.
> Module default I2C address is 0X69
> For testing used [this](https://github.com/junichikatsu/CDM7160)

Ingrown29, 2019-03-09:
Hi guys, I have the same sensor and I want to make it work but when I am trying to compile the code on Arduino IDE (Arduino UNO) I get this problem:

\Arduino\libraries\Wio_LTE_for_Arduino\src/Internal/AtSerial.h:5:18: fatal error: string: No such file or directory

Do you know where is the problem? I am using the code from link which TD-er posted.
Thank you for every little help

Egyras, 2019-10-10:
@Ingrown29  just remove Wio_LTE library from the code as you dont need it.

Egyras, 2021-01-25:
> That's right, it isn't. Was just joking. I'll have to order one myself too, in order to support it.

Hi, did you any tests with CDM7160 ? I can see that its in playground:
https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P176_CDM7160.ino
Who could compile for testing and provide link to download .bin file ? Could do proper testing myself.

tonhuisman, 2021-01-25:
I'll give it a go, what type of ESP unit are you using? (I'll post just 1 .bin file)

Egyras, 2021-01-25:
> I'll give it a go, what type of ESP unit are you using? (I'll post just 1 .bin file)

Wemos d1 mini (4MB)

tonhuisman, 2021-01-25:
This is a 'test' build, based on current mega branch (where Releases are built from) with P176 from PluginPlayground added and fixed to compile with current state of development.
Done no optimizations yet (especially memory use could use some attention, I think, and allowing the ESP to run normally when the hardware isn't connected), that's something to be done once it proves to be working fine.
Uploaded to a test unit and verified the plugin is available and configurable.

[ESP_Easy_mega_20210125_test_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5869223/ESP_Easy_mega_20210125_test_ESP8266_4M1M_VCC.zip)

@Egyras Please report your findings here.

Egyras, 2021-01-26:
> This is a 'test' build, based on current mega branch (where Releases are built from) with P176 from PluginPlayground added and fixed to compile with current state of development.
> Done no [...]

Thanks, stared testing yesterday late evening. All looks good, did quick re-calibration with outside air (shorting for a few seconds pin 7 on cdm7160 to gnd). Memory usage looks fine to me aa well.
Herei is graphs:
http://188.69.132.115:3000/d/vSL83CiRz/comfoair-ca-350-advanced-dashboard?viewPanel=9&orgId=1&refresh=10s&from=1611642782036&to=1611644582036

tonhuisman, 2021-01-26:
> Memory usage looks fine to me as well.

Didn't do any measurements yet, but looking at the code parts of it is always in memory, even when the plugin isn't configured.
I'll do some changes and measurements on memory usage in the coming days/weeks, (without a sensor as I don't own one). It'll add the option of running multiple sensors on the same ESP (by using a I2C multiplexer) without interfering each other, software-wise.

TD-er, 2021-01-26:
> Thanks, stared testing yesterday late evening. All looks good, did quick re-calibration with outside air (shorting for a few seconds pin 7 on cdm7160 to gnd). Memory usage looks fine to me as well.

Just FYI:
It is always tricky to perform a forced base calibration on CO2 sensors (telling the sensor what is 400 ppm)
Especially outside after sunset, because the plants then start to exhail CO2 and thus increasing the CO2 concentration.

For example the CO2 concentration measured by my car outside of the last 4 days:
[image]

When there is no wind, the CO2 concentration outside can get as high as 700 ppm.
So if you then tell the sensor the current CO2 concentration is 400 ppm, you will introduce an offset and probably a flat curve of 400 ppm when you reach values lower than the concentration when you set the 400 ppm value.

Egyras, 2021-01-26:
> Just FYI:
> It is always tricky to perform a forced base calibration on CO2 sensors (telling the sensor what is 400 ppm)
> Especially outside after sunset, because the plants then start to exhail CO2 [...]

I believe in winter full of snow plants shouldnt do any influence, btw thanks for tips. Can try to do some checks within chamber and some 100% nitrogen, just need to source it :)

TD-er, 2021-01-26:
100% nitrogen is not helping to set the 400 ppm level.
Maybe you're setting the 0 ppm value, but I doubt that's a value you can set.

Egyras, 2021-01-26:
> 100% nitrogen is not helping to set the 400 ppm level.
> Maybe you're setting the 0 ppm value, but I doubt that's a value you can set.

you can with same pin 7 to gnd if holding longer then 12 seconds with this CDM7160 ;)

TD-er, 2021-01-26:
I would not try to change it.
It should be set by the manufacturer and I don't think that value will drift a lot during use.
Better try to take a proper reading during the day in a well ventilated room (not in industrial area of close to a busy road) and make sure no sunlight will hit the sensor.
It is really easy to mess up the calibration of a CO2 sensor, so I would not try to change the 0 ppm value as you will very likely ruin the sensor.

Egyras, 2021-01-26:
> I would not try to change it.
> It should be set by the manufacturer and I don't think that value will drift a lot during use.
> Better try to take a proper reading during the day in a well ventilated [...]

I live outside city, so enviroment around is clean without busy roads and etc. Tried to reset a few times before and seems it doesnt drift much from 400 ppm value if I leave outside for longer time.

tonhuisman, 2021-12-31:
I have now created a PR to add this plugin to ESPEasy.
Is anyone available to test the cleaned up implementation? If needed I can supply a build once the Github Actions run is ready. (I've added it ti the TEST_D set for now.)

fbloemhof, 2022-01-03:
I have such a sensor and would like to test it, but I don't have enough experience with ESPEasy to understand what you mean by the TEST_D set.

tonhuisman, 2022-01-03:
You can download the `Binaries.zip` file from the [Github Actions page](https://github.com/letscontrolit/ESPEasy/actions/runs/1644338052) (that's for the last commit I made, new builds will get a different page).
From that zip file, extract the build you need for your ESP (ESP8266 or ESP32) with `TEST_D` in the name, and flash the .bin file to your ESP (can be via OTA update if it is already running ESPEasy).
Then you can add a task on the Devices page with this plugin: `Gases - CO2 CDM7160 ...`, physically connect and then enable that task. Now you should start to see the CO2 value as measured by the sensor (_might_ take some time, like minutes, before it actually shows anything, as it may need some warm- or start-up time, that IDK).

tonhuisman, 2022-01-10:
@fbloemhof Did you get it going, or do you need some more support in getting the software on the ESP?

fbloemhof, 2022-01-12:
@tonhuisman I think I have it running and it is actually measuring CO2!

Will do some further testing and need to find out how I get the data in my Home Assistant instance so I can compare it easily to another CO2 sensor.

fbloemhof, 2022-01-12:
@tonhuisman I noticed that it measures a value of 16383 when starting up, then it takes around 30 secs to get to the actual measured value.

It is running for now and I have the value visible in HA so will leave it now for a couple of days. 

Sidenote, it is running on 3,3V (for now this was the most simple setup) whilst the documentation states that is not possible...

tonhuisman, 2022-01-12:
> it measures a value of 16383 when starting up,

Hm, maybe we could add a check for that, to return `NaN` instead of that unrealistic value, but currently there is no check.

> then it takes around 30 secs to get to the actual measured value.

That is quite common for this type of sensors, they need to warm up (literally) to get into their optimal measurement state.

TD-er, 2022-01-12:
The MH-Z19 sensor has something similar.
I think it outputs something like 15000 to indicate it rebooted.

TD-er, 2022-01-12:
> Sidenote, it is running on 3,3V (for now this was the most simple setup) whilst the documentation states that is not possible...

Some other CO2 sensors do seem to run longer per sample when operating on a lower voltage.
Maybe this sensor does too, or if not it may appear more noisy or show an offset.
Would be interesting to check.
N.B. this offset can be a scalar, or a fixed offset, or both.

fbloemhof, 2022-01-13:
After ~24 hours running all still seems well.

tonhuisman, 2022-01-13:
I have added a check to not send out the data if the CO2 level is > 15000. There was already a warning when the level is > 10000, that is still in place.
The Github action builds fail because there is 1 that results in a too large .bin file, so I've created local builds for ESP8266 and ESP32 of the `TEST_D` taste, I assume you are using:
ESP8266:
[ESP_Easy_mega_20220113_test_D_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7866048/ESP_Easy_mega_20220113_test_D_ESP8266_4M1M.zip)
ESP32:
[ESP_Easy_mega_20220113_test_D_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7866087/ESP_Easy_mega_20220113_test_D_ESP32_4M316k.zip)

fbloemhof, 2022-01-18:
Good assumption.

Finally had the time to flash. First results look good.

fbloemhof, 2022-01-21:
It is still going strong. Having it running side-by-side to an MHZ19 sensor they show almost the exact same results so I'm guessing that running on 3,3 V is also good.

PR #3898 (2022-03-31): [P127] CDM7160 I2C Co2 sensor plugin
Resolves #986 

New plugin for the CDM7160 I2C CO2 sensor.
Added to the TEST_D build set.

TODO:
- [x] Documentation
- [x] Testing (help requested, as I don't own this sensor, it is rather expensive, compared to other similar sensors) ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/986#issuecomment-1018299910))
closed 2022-03-31 by TD-er