letscontrolit/ESPEasy #987: ESP32: flashSerial.cmd script translation to bash as flashSerial.sh

N0rbert, 2018-03-02:
### Steps to reproduce
1. Download latest ESPEasy32 build from [its web-page](https://www.letscontrolit.com/wiki/index.php/ESPEasy32).
``` 
wget http://www.letscontrolit.com/downloads/ESPEasy32_R20100.zip
unzip ESPEasy32_R20100.zip
cd ESP32flash
```
2. Try to find shell script to burn firmware on GNU/Linux

### Actual behavior
Today there is no such script.

### Resolution
I wrote my own script `flashSerial.sh` and share it with ESPEasy community. It is straightforward as original windows one:

```
cat flashSerial.sh
#!/bin/sh
echo -e "On GNU/Linux you can install esptool with the following command:"
echo -e "\n\tsudo pip install esptool --upgrade\n"

read -p "Comport (example /dev/ttyUSB0): " comport
echo $comport

read -p "Build (example 20000, ..): " build

echo Using com port: $comport
echo Using bin file: ESPEasy32_R$build.bin

esptool.py --chip esp32 --port $comport --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader.bin 0x10000 ESPEasy32_R$build.bin 0x8000 ESPEasy.ino.partitions.bin

echo -e "\n"
read -p "Press any key to exit ..." -n 1
```

Then make it executable and run:
```
chmod +x flashSerial.sh
./flashSerial.sh
```

Hope this helps.

TD-er, 2018-03-02:
Please remove the "cat" command from the first line in the code. :)
And maybe you can add a check to use a parameter as first suggestion for the comport?

For example:
```
if [ $# -gt 0 ]; then
  comport=$1
else
  read -p "Comport (example /dev/ttyUSB0): " comport
fi
```

N0rbert, 2018-03-02:
Thank you, **TD-er**. I updated my script.

Barracuda09, 2018-03-02:
Now you still need to hit enter if you specify an argument, running this script.

N0rbert, 2018-03-02:
@Barracuda09 yes, and I'm comfortable with this.

Barracuda09, 2018-03-02:
Well for automated build,  I find it annoying to press to many buttons ;)

If you use @TD-er  approach you have both worlds combined :)

closed 2023-09-04 by TD-er