letscontrolit/ESPEasy #989: FR: Hysteresis, upper and lower limit for every sensor value

M0ebiu5, 2018-03-02:
I would like to request the option to enter hysteresis, upper and lower limit for each sensor value.

This would reduce network traffic, lower db load and avoid unrealistic spikes.

TD-er, 2018-03-02:
So if the measured value is within the hysteresis, no new value is reported?
That means also the last sent value must be stored.
Should there also be some timeout by which time a new value must be reported? For example Domoticz will mark items red when no new value was reported in a while.
Should this maximum time between reports be a somewhat "global" setting of the controller, or the sensor?
I assume the hysteresis should be part of the sensor settings?

And is this hysteresis an absolute deviation, or relative? Upper and lower the same?

M0ebiu5, 2018-03-02:
> So if the measured value is within the hysteresis, no new value is reported?

yes

> That means also the last sent value must be stored.

yes

> Should there also be some timeout by which time a new value must be reported? For example 
Domoticz will mark items red when no new value was reported in a while.

good point - could be useful

> Should this maximum time between reports be a somewhat "global" setting of the controller, or the sensor?

global setting, not for every sensor

> I assume the hysteresis should be part of the sensor settings?

yes - in the section, where one can also enter the formula - but a separate section is also possible

> And is this hysteresis an absolute deviation, or relative? Upper and lower the same?

upper and lower absolute
hysteresis would be nice to have % and absolute (if not, only absolute)