letscontrolit/ESPEasy #993: Add %unit% variable in MQTT subscribe topic

kluszczyn, 2018-03-02:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Change default subscribe MQTT topics (tested with OpenHAB controller) from /%sysname%/# to /%unit%/#. Variable %unit% will not be resolved to a unit number.

### Expected behavior
Parse variable %unit%  in MQTT _subscribe topic_. 

### Actual behavior
Currently, only %sysname% in MQTT _subscribe topic_ is parsed into variable value.
Other sys variables are not parsed. 

Parsing %unit% variable (or maybe other sys variables also, like %ip% or %mac%) would add flexibility for manipulating MQTT messages in controllers like OpenHAB. Changing unit name would not require changes in OpenHAB items definitions.

### System configuration

Hardware:
ESP12F

Software or git version:
ESPEasy_mega-20180224 precompiled image

TD-er, 2018-03-02:
%unit% , %mac%, %mac_int% and %ip% are already present in the list of usable system variables.
N.B. %mac_int% = Last 24 bit of MAC address as integer, to be used in rules.

I think %unit% is one of the last ones added, so maybe test a later build.
The build of the 24th is sooo last-week ;)

If it is not parsed in MQTT topics, then that's a bug.

kluszczyn, 2018-03-02:
Sys variables are parsed in MQTT **_publish_** topics, but not in **_subscribe_** topic.

TD-er, 2018-03-02:
OK, that's a bug.

kluszczyn, 2018-03-02:
As far as I understand code, MQTT subscribe topic is defined in Controller.ino where %sysname% is parsed (and only this sys var)
https://github.com/letscontrolit/ESPEasy/blob/635c149f129fd6dd9ff7bab046335284c3b3b4ca/src/Controller.ino#L164-L169

This "hardcoded" parsing is also applicable to LWT topic:
https://github.com/letscontrolit/ESPEasy/blob/635c149f129fd6dd9ff7bab046335284c3b3b4ca/src/Controller.ino#L139-L142

and also to MQTTStatus function:
https://github.com/letscontrolit/ESPEasy/blob/635c149f129fd6dd9ff7bab046335284c3b3b4ca/src/Controller.ino#L234-L247

kluszczyn, 2018-03-02:
Quick and dirty solution would be to add after all instances of:
`.replace(F("%sysname%"), Settings.Name); `
this one:
`.replace(F("%unit%"), Settings.Unit); `

But I did not test it.

TD-er, 2018-03-02:
Better use `parseSystemVariables(pubname, false)`
See its declaration at the StringConverter.ino file.

kluszczyn, 2018-03-02:
Using function `parseSystemVariables(pubname, false)` in MQTT **subscribe** topic would allow using all sys variables, like `%syssec%`, resolved at espeasy startup to unpredicted/random values :), But we would have great flexibility!

TD-er, 2018-03-02:
Yeah, we could add some check for "stable" variables there.

closed 2018-03-09 by TD-er

kluszczyn, 2018-03-14:
Hi,
two more places has left for fixing:
https://github.com/letscontrolit/ESPEasy/blob/c94e5cb90354e5ce40933ab9061ac3eaceeeb756/src/Controller.ino#L138
and
https://github.com/letscontrolit/ESPEasy/blob/c94e5cb90354e5ce40933ab9061ac3eaceeeb756/src/Controller.ino#L247

TD-er, 2018-03-15:
Added those suggestions

greenoid, 2018-10-03:
/%sysname%_%unit%/
comes in very handy in OpenHAB.
Results in "ESP_Easy_10" for unit number 10. Easy.