letscontrolit/ESPEasy #1003: Convert MQTT import wind direction to text for display.

mattlward, 2018-03-03:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
There are no real steps other than MQTT cannot import text.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
I would like to be able to display directional data in text for wind.

### Actual behavior
I can only display direction in degrees on OLED

### System configuration

Hardware:
Wemos D1's and several 1603 and 1102 displays.
Software or git version:
Several latest versions and a couple of custom versions.  I generally stay current on nightly builds of 2.0

Thanks Matt

TD-er, 2018-03-03:
Is it OK to do this conversion similar to a system variable?
So `%bearing%(value)`, where value is the int representation of the direction in degree.

mattlward, 2018-03-03:
I think that conversion would be fine. as long as %bearing% could be assigned via MQTT.

TD-er, 2018-03-04:
Can you give an example of a MQTT message given to ESPeasy?

mattlward, 2018-03-04:
I set it as a simple broadcast...  I publish WindDeg/(0-360) once every 3 minutes.

TD-er, 2018-03-04:
And the MQTT importer gets that? So in a rule you can access is as if it is a sensor value, but only from the MQTT importer?

mkomarac, 2018-03-04:
I am looking just for that. 
I don't use ESPEasy put request is the same.
My MQTT broker gets this data (read from MQTT Spy):
{
          "battery": "OK",
          "channel": 1,
          "id": 234,
          "mic": "CHECKSUM",
          "model": "AlectoV1 Wind Sensor",
          "time": "2018-03-04 08:24:02",
          "wind_direction": 135,
          "wind_gust": 0,
          "wind_speed": 0
}

And I would also like to see wind direction data in this case SE format.
I am using HASSio

thank everyone for help!

TD-er, 2018-03-04:
I made the change a little bit more elaborate and created a framework to allow to add other conversions when needed.
For example Celsius => Fahrenheit could be added.

mattlward, 2018-03-06:
So, this is live?  What is the best way to set a variable without mqtt?

TD-er, 2018-03-06:
You should be able to use it anywhere.
Use `%conv_wind_dir%(value)`
instead of 'value', use any int or float value, like  `%conv_wind_dir%(123)`
N.B. the output of a sensor can also be used, like `[weather#winddir]`
That would look like  `%conv_wind_dir%([weather#winddir])`
You can use it for example in the OLED Framed plugin or in rules.

mattlward, 2018-03-06:
@TD-er, I am struggling with giving up 6 MQTT imports.  Can I use http to populate dummy values and then use them with the OLED?  You know, it is the old limit of 1 mqtt import task.  Weather display needs to have wind speed, wind gust speed, wind direction, temperature, humidity and heat index or "feelslike".  I am going to try to use global sync once one unit has all of the values, but I understand that I will have to have task number uniformity for that to work, so it will take a bit of effort to establish that.  

Sorry to be a pain here, I want to move forward in versions, but after 228, 2 mqtt importers do not work well and I need to get away from that mode of operation.

Grovkillen, 2018-03-06:
@mattlward you can use event,EventName=<floatvalue> to populate dummy device values. Works with mqtt as well as http.

mattlward, 2018-03-08:
I have managed to get completely away from MQTT weather import now!  But, I have a question...  I would like the display to be able to display the wind direction and MPH on 1 line in the OLED task... But, there is not enough room on the task line for the following:
%conv_wind_dir%([D2#WindDeg]) @ [D1#Wind] mph

Just can't type it all in, any ideas?
BTW, I now love global sync!

TD-er, 2018-03-08:
I could change the conversion argument and you can shorten the names of the variables.
So this would be possible:
`%conv_wdir%([D2#WD]) @[D1#W]mph`
Could that be workable?

mattlward, 2018-03-08:
Will see if it fits.

TD-er, 2018-03-08:
It is 32 characters, which should fit. But then I still have to change the conversion name.

TD-er, 2018-03-08:
I merged a fix with even shorter strings. See #1033 
Should be included in the next build tomorrow.

TD-er, 2018-03-15:
This can be closed?

mattlward, 2018-03-15:
I think so.  works great

closed 2018-03-15 by TD-er