letscontrolit/ESPEasy #998: Sonoff Touch - network problem

dzban1976, 2018-03-03:
[]([https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4943&sid=951594c18d01a65b8e02c7113c3935dd])
is it normal that the button does not work when sonoff loses network connection ?!
example:
my router wifi shutdown -> sonoff touch not connected ->I touch the button on the device -> no reaction
why ?!

TD-er, 2018-03-03:
It should not be. What build are you using?

dzban1976, 2018-03-03:
ESP_Easy_v2.0-20180228_dev_ESP8285_1024
ESP_Easy_mega-20180302_dev_ESP8285_1024

TD-er, 2018-03-03:
So in short, you have set the button to act on the relais in the Sonoff.
I guess you did this via rules?
Can you post these?

Could you also try to set the "Message Delay" to 100 (default = 1000)
This is located at Tools => Advanced => Controller Settings.

dzban1976, 2018-03-03:
rules 1:
```
on touch#state do
  if [touch#state]=1
   gpio,12,1
  else
   gpio,12,0
  endif
 endon 
```
rules2:
```
on relay#state do
  if [relay#state]=1 
   gpio,13,1
  else
   gpio,13,0
  endif
 endon 
```
I can not change anything now...
the device is not responding again
Espoler (tx,rx console)
```

INIT : Booting version: mega-20180224
INIT : Warm boot #136
FS   : Mounting...
FS   : Mount successful, used 76806 bytes of 113201
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:22096
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be sonoff_43 with address 192.168.4.1
IP   : Static IP :234924224
WIFI : Connecting Domek_saloon attempt #1

Exception (3):
epc1=0x402691cf epc2=0x00000000 epc3=0x00000000 excvaddr=0x40000003 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  00000005 00000000 00000020 40100e2e  
3ffffe10:  3ffeb600 00000101 4026c10f 3ffffed0  
3ffffe20:  3fff05b0 3ffeff78 3fff06d0 00000014  
3ffffe30:  4026c521 3ffffed0 3fff0100 3ffeff78  
3ffffe40:  3ffed82c 00000001 4026c56d 3ffffed0  
3ffffe50:  00000000 00234e93 40102112 00000100  
3ffffe60:  7fffffff 3ffebef0 3ffebef0 00000001  
3ffffe70:  40268c79 00080000 3fffc200 00000022  
3ffffe80:  ffffffbb 3ffed82c ffffffbb 3ffed83c  
3ffffe90:  3ffef0d0 0000001b 00000000 4026d2cc  
3ffffea0:  00000000 3fff69cc 3ffed82c 00000000  
3ffffeb0:  00000000 00000030 00000014 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000000 00310303 00640100 0000004b  
3ffffee0:  3ffed850 00000116 3ffed86b 3ffed844  
3ffffef0:  00000000 3ffed850 3ffed85e 3ffed88a  
3fffff00:  00000000 00000000 3ffed874 3ffed8c4  
3fffff10:  3ffed890 3ffed8ac 00000000 3ffed925  
3fffff20:  00000000 00000000 0000001b 00000000  
3fffff30:  3fff6d8c 4026cd97 3ffef0d0 3fff69cc  
3fffff40:  00000000 00000000 3ffef0d0 3ffed82c  
3fffff50:  00000000 0000013e 00000000 0000001b  
3fffff60:  00000000 0000013e 4025250b 3ffef0d0  
3fffff70:  3ffed820 3fffdcc0 3ffeb6d8 3ffeb6d8  
[... 64 lines cut ...]
fe c0 3c 20 05 89 06 45 d1 ff 16 4c 18 32 2c 00 d2 2c 15 e2 0c 26 9c 03 8c cd 28 23 8c a2 d0 2d 
20 05 7a 06 86 00 00 00 dd 03 f8 6c 52 cc 18 c6 07 00 68 4f 2d 05 3d 0f 59 01 69 11 85 89 06 31 
1f fe 2d 0f 0c 04 01 72 d2 c0 00 00 68 11 58 01 fd 06 56 cf fd f1 d9 ff 1c 43 2d 0f 01 b4 d8 c0 
00 00 2d 0f 32 cc 34 42 a0 14 01 b1 d8 c0 00 00 16 cd 0f 28 2d 16 72 0f 48 52 16 24 0f 80 3e 01 
2d 0d 30 38 31 c0 04 00 06 39 00 00 00 21 42 fe c0 3c 20 45 80 06 45 c8 ff 16 5c 0f 32 2c 00 d2 
2c 15 16 a3 07 16 fd 00 28 23 8c c2 28 2d 8c 82 2d 0d 05 71 06 46 00 00 dd 03 2d 0c 05 df ff ed 
02 16 b2 05 f2 2c 13 f0 f0 04 16 4f 04 21 10 ff 0c 0f 38 02 06 0d 00 00 42 13 10 22 1e 10 27 14 
02 c6 24 00 86 25 00 00 00 29 3f 06 01 00 41 08 ff 29 04 68 33 2d 03 31 f1 fd 0c 04 69 11 01 44 
d2 c0 00 00 68 11 3d 0f fd 03 3d 06 56 83 fc 46 03 00 f0 3f 20 85 c8 02 e0 2e 20 3d 0f 45 c9 02 
e8 6c f2 cc 18 06 07 00 68 4e 3d 0e 2d 0f 62 61 01 45 7a 06 31 e2 fd 2d 0e 0c 04 01 35 d2 c0 00 
00 68 11 ed 06 56 fe fd e1 9e ff 1c 43 2d 0e 01 77 d8 c0 00 00 2d 0e 32 cc 34 1c 44 01 75 d8 c0 
00 00 28 2d 8c 82 38 62 8c 43 d0 2d 20 c0 03 00 2d 0c 05 df ff 46 06 00 68 33 86 e2 ff 00 30 23 
20 32 61 00 05 c4 02 32 21 00 28 33 56 9f f5 c6 d6 ff 08 b1 c8 a1 d8 91 e8 81 f8 71 12 c1 30 0d 
f0 00 00 00 12 c1 d0 d9 91 7d 03 dd 02 0c 13 0c 02 d0 23 83 09 b1 c9 a1 e9 81 f9 71 20 20 74 40 
60 f4 56 b2 0a 70 23 83 56 52 0a 16 26 0a e2 2d 01 cd 06 22 1e 39 67 b2 02 20 c2 20 22 1e 1e c0 
c0 f4 f0 22 11 c7 a2 02 20 c0 f4 0c 4f 52 2d 13 2d 0e 50 5f 10 3d 07 4d 0c 9c 15 0c 15 69 11 79 
01 c5 67 03 20 20 74 68 11 78 01 06 04 00 62 61 01 72 61 00 85 66 03 72 21 00 68 11 20 20 74 80 
32 01 30 58 31 c0 41 41 1b 55 50 c4 83 b6 2c 05 30 38 31 26 03 b6 fd 02 fc 82 28 7d ca 77 c0 c6 
c0 79 12 c2 52 04 e0 2e 20 45 af 03 42 2d 13 0c 43 20 f0 74 47 03 1c 28 7d 22 12 04 dc 42 0c 12 
22 4d 48 4d 0d 1c a2 01 bc e3 c0 00 00 c6 00 00 00 f2 a0 f4 08 b1 2d 0f c8 a1 d8 91 e8 81 f8 71 
12 c1 30 0d f0 00 00 00 12 c1 f0 09 31 5d 02 30 40 74 9c 62 22 22 15 8c a2 38 15 2d 05 85 4f ff 
06 02 00 00 00 38 15 2d 05 85 58 ff 08 31 12 c1 10 0d f0 00 35 ad fe 3f 12 c1 e0 c9 61 0c 0c 09 
71 c9 01 c7 12 29 10 31 20 c5 62 06 c7 92 0c 21 f9 ff 01 56 d1 c0 00 00 đŚŽ

```

TD-er, 2018-03-03:
Did it work before?
This looks like a crash/reboot loop.

closed 2018-03-04 by TD-er

Grovkillen, 2018-03-05:
#484 is related to this issue.