letscontrolit/ESPEasy #1007: Write user-specific data in a separate file

micropet, 2018-03-04:
I would suggest to outsource the user-specific data from the file ESPEasy.ino
to a separate file.

This makes an update much easier.

TD-er, 2018-03-04:
You mean default WiFi credentials and such?

micropet, 2018-03-04:
Yes. From 
// --- Basic Config Settings -
to
//add this if you want SD support (add 10k flash)
//#define FEATURE_SD

micropet, 2018-03-04:
I think you make a "user_config.txt" and we use a "user_config.h" or similar. If we copy the source locally we can not override it.

TD-er, 2018-03-04:
So perhaps something like a separate .ino file that will override the defaults and only if defined (thus present).
And this file must be in the .gitignore to prevent uploading to Git.
Also a sample must be supplied.
I guess this allows to set a lot more settings, but then a save must also be supplied.

Somewhat related issues:
#957  #872  #512

micropet, 2018-03-04:
Yes, these are the right ideas. Seems a bit more complicated than I thought. :)

TD-er, 2018-03-19:
See comments below this PR https://github.com/letscontrolit/ESPEasy/pull/1116
And #807, which is a way better solution. That's why I reverted my own changes.

closed 2018-03-19 by TD-er