letscontrolit/ESPEasy #1010: NTP should trigger an event to update outputs regardless of their current state.

TD-er, 2018-03-04:
See discussion on [Forum](https://www.letscontrolit.com/forum/viewtopic.php?p=25471#p25471)

> However the Clock output issue still remains.
> How to reproduce:
> Put Clock output plugin within a time schedule ON period.
> Remove power to ESP, replace power to ESP.
> Clock output is now OFF, and will remain OFF till next ON period comes round.

What should happen:
On reinstating of power, Clock Output should compare current time to setvalue time/status and switch output accordingly.

TD-er, 2018-03-08:
Added a number of new events in PR #1032 
Among these is one for time update.

sujitrp, 2018-03-14:
Can this be closed by this solution ?
http://raspberrypi.tomasgreno.cz/ntp-client-and-server.html

TD-er, 2018-03-15:
@sujitrp Nope, that's just installing a local NTP server.
The thing is (as I understood it correctly), that there may be some delay between power on and first initial NTP lookup.
If the rules already fired and set some pins, they should either correct that when the time is set, or react on an event triggered by a clockset action from a successful NTP query.
Since PR #1032 , on first successful NTP query, an event is sent, so that may help in applying time based rules.

Grovkillen, 2018-03-16:
https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#.28System.29_events

These are now documented (I had missed to document the Time# events).

closed 2018-03-16 by Grovkillen