letscontrolit/ESPEasy #1012: Web interface doesn't work on iOS

ax42, 2018-03-04:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Try to open the web interface from Safari on iOS using either the IP or hostname of the device.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
Web interface should open in the browser.

### Actual behavior
Tell us what happens instead?
Connection starts, sometimes one can quickly see the interface flash by, and Safari gives the error "Safari cannot open the page because the network connection was lost".

This only started happening after updating to the 20180228 build.  Works fine from Safari on Mac.  Fails on 2 different iOS devices for me (iPhone and iPad), both of which worked before.

### System configuration

Hardware:

Wemos D1

Software or git version:

Build | 20100 - Mega (core 2_3_0)
GIT version | mega-20180228

TD-er, 2018-03-04:
You updated to the 20180228 build, but what was the previous build before that?

It may be related to the "2k webserver" update we did.
There were some other problems with the /json url also.
Can you check with build 20180228 and with one of the last builds whether the /json url does return some data?

ax42, 2018-03-04:
As per https://github.com/letscontrolit/ESPEasy/issues/968 looks like my previous release was 0224 (I've only just started using ESPEasy).

The 0228 build returns the same error as described above when retrieving `/json` on iOS.  Works fine on Mac.

0304 returns correct-looking json when fetching `/json` on either iOS or Mac, however when retrieving any other URL from iOS (eg `/` or `/tools`) I get the same error as above.

TD-er, 2018-03-04:
OK, then it looks like Apple's OS is looking at the headers much more strict than the other browsers.
The Content-Length parameter in the header is inadvertently set to 0, while it is even optional and not usable in chunked transfers.

Could you test using a non-Apple browser like Firefox/Chromium/Opera ?

andibaciu, 2018-03-06:
I can confirm entire story tell by ax42 ... there are problems with loading page on Safari from iOS, starting from version 0228 to now.
Safari gives the error "Safari cannot open the page because the network connection was lost" after a quickly show of the page ...
Now i don't have access to install other browser on iOS device to test that, but soon i will try this and tell results.

TD-er, 2018-03-06:
We have to look into that ASAP, to disable these headers for the chunked transfers.
From the HTTP library in the core libs, these headers should be disabled (or properly set)

- Content-Length
- Content-Type

andibaciu, 2018-03-06:
I install Firefox and check to see the page is loading... the page is loading and see all the data but there is a big problem with zoom in or zoom out... i cannot see data arranged correctly when is in portrait mode and the page with devices is not scaled correctly on entire landscape page ...
Do you have other test to do ?

TD-er, 2018-03-07:
It should be fixed. But I don't have anything from Apple to test.

Also improved the response time of the webserver by making it send smaller pieces at once when large strings were sent to the webserver. (e.g. a 3+ kB CSS stylesheet)

beicnet, 2018-03-07:
@TD-er You can test it [here](https://www.browserling.com/)

TD-er, 2018-03-07:
@beicnet Hmm, looks like I have to add portforwarding then to make it accessible from the outside.

beicnet, 2018-03-07:
@TD-er That's correct

TD-er, 2018-03-07:
Just tested and apparently there is somewhere a firewall or something blocking, since I cannot load the page from any of the browsers there.

beicnet, 2018-03-07:
:disappointed:

andibaciu, 2018-03-08:
Hello,
After update to 2018.03.07 the page is displayed on Safari but I ask if everything is ok with the menu... now menu are displayed one below each other and not side by side like o other devices. Menu are displayed with this problem on all kind of browser on iphone devices.

Grovkillen, 2018-03-08:
That is the quick fix introduced some days ago. A permanent fix is not yet online.

TD-er, 2018-03-08:
Indeed, if the view is less than 960 pixels wide, the menu items are displayed vertical aligned.
But good to hear the other patch is also working, which means we can output json via the streaming solution again.
That will save memory.

TD-er, 2018-03-15:
This issue can be closed.
Please re-open when there is still an issue.

closed 2018-03-15 by TD-er