letscontrolit/ESPEasy #1015: Add other simple conversions like Celsius to Fahrenheit, etc.

TD-er, 2018-03-04:
See PR #1014 for inspiration.

Please add some suggestions for other conversions.
Current ideas:

- Celsius => Fahrenheit
- m/s => Beaufort
- metric length => imperial length

Budman1758, 2018-03-05:
24 hour clock time to 12 hour AM, PM

Grovkillen, 2018-03-08:
Minutes to days (for uptime).

closed 2018-03-08 by TD-er

Grovkillen, 2018-03-09:
FYI: https://www.letscontrolit.com/wiki/index.php/EasyFormula#Conversions

Grovkillen, 2018-03-09:
The 24h >>> 12AM/PM was never included in this commit.

TD-er, 2018-03-09:
Should it be included?
I've still to convert some parts to floats, as suggested by @Grovkillen documentation ;)

Grovkillen, 2018-03-09:
Yes I think it's needed for many users.

TD-er, 2018-03-09:
But who uses am/pm? I always wonder this when searching for the setting to change it to 24h mode ;)

Grovkillen, 2018-03-09:
Many US users... "Imperial" says it all :wink:

Budman1758, 2018-03-09:
Its not absolutely necessary.  Just a convenience for us Yankees.  :)

TD-er, 2018-03-09:
It's already building here... ;)

TD-er, 2018-03-09:
Done :)

Now cherrypick to v2.0

Grovkillen, 2018-03-09:
AM/PM is supported with a new system variable :

https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables