letscontrolit/ESPEasy #1016: Webserver- responsive design for mobile devices.

s0170071, 2018-03-05:
Displaying the ESPEasy webpages on a mobile phone is ... I mean it just doesn't work so good for me. And I am not even using a very old one. Have there been thoughts on tweaking the webpages towards "responsive design", so that they can be used on mobile devices more easily ? 

This may be related: #906
@doxikus: do you have plans for that ?

TD-er, 2018-03-05:
I have plans  to add json rest interface, to allow for a more responsive site.
Let the browser do the heavy work, not the esp

s0170071, 2018-03-05:
Whats the timeline for this ? Would it make sense to put work into this in the meantime ?

doxikus, 2018-03-05:
Hi,
Later I will post longer response, in short I have done some work on this topic, I think that I will create pull request this week, skin is almost ready, I have done with updating markup now I'm on css styling.

I have fetched all changes from remote upstream and merged inside my branch, no conflicts, still there is some issues that I need to fix.

TD-er, 2018-03-05:
@s0170071 The timeline for Json REST API is unknown yet.
It involves a lot of changes, so I've been putting it on hold.
I plan to use [cereal](http://uscilab.github.io/cereal/) to do this. But that's another issue. Not sure if I added the issue already. If not, I will add it. It has been discussed on the Slack.

doxikus, 2018-03-05:
Oh, I have open forum and find invitation from psy0rz to slack channel :-). If you are ok with this, we can keep separated branch for new skin/theme or use my forked repo, invite few people to try FW and test on devices. There is tricky part because I have use small part of .js for menu and some other visual enhancements. Js is hosted like css on SPIFFS, so it needs to be uploaded manually.

mattlward, 2018-03-13:
@TD-er, I assume some the changes have been made.  The mobile interface is much faster now on the latest nightly and now the top menu choices do not overlap and break.  Much better!

TD-er, 2018-03-13:
Yep, @s0170071 introduced a new way of sending data to the browser, which takes much less resources from the ESP.
And last weekend I made a few small enhancements to reduce the memory usage even more.
@mvdbro made some changes in handling the rules, which only improved rules handling with a factor 10(!!!!)
So all-in-all you can say there have been some improvements indeed :)
@doxikus did some changes on the CSS to make it usable on mobile devices.

It's a team-effort :)

Budman1758, 2018-03-13:
@doxikus I would be happy to help test it out. I have some units I can play with.

doxikus, 2018-03-13:
HI @Budman1758 thnaks in advance, I have small demo on https://github.com/doxikus/ESPeasyskin download repo and open in browser, index.html or tools.html. Still need to update some things but very soon I will have demo firmware so anyone can try it on demo device.

TD-er, 2019-11-04:
The current implementation is responsive.
So I will close this one.

closed 2019-11-04 by TD-er