letscontrolit/ESPEasy #1017: MODBUS support

s0170071, 2018-03-05:
I have conducted some experiments on my SMA STP5000TL solar inverter and found that it can be probed easily using MODBUS /TCP. See this entry in the forum
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4960&p=25485&hilit=sma#p25485
While I was just quick and dirty assembling the frame, there could be a more general aproach to support modbus.

LeeNX, 2018-03-05:
I have some Estron power meters and RS485 serial interface devices, which could come handy if there is anything that is not too hard to put in place on a NodeCMU device for testing. Will try and keep an eye on this issue.

TD-er, 2018-08-10:
Not yet.
Recently I have been brushing up my Modbus code a bit, which I wrote about a year ago for the new Senseair sensors (K33 et.al.) See https://github.com/TD-er/ESPEasy/tree/feature/MODBUS_enhancements
The current Modbus in ESPeasy main repo is more about RTU over ethernet and the code I wrote for those Senseair modules is about RTU over UART.
I still have to find a way to merge those into a good interface.

But to be honest, I want to improve stability of the existing code first to get a release out, which is really way overdue.
So first:

- Improve handling of controllers (currently working on)
- Have a look at the unstable HTTP-controllers.
- Proper handling of GPIO pins (switch does act a bit strange every now and then, not sure why)

There is some support in the playground if I am not mistaken for the Eastron DIN meters.
https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P150_SDM120C.ino
Not sure if those are the Modbus or the M-bus versions of those meters.

s0170071, 2018-08-10:
+1 for the stable release :-P

TD-er, 2018-08-11:
If you find something that wasn't obvious connecting the Eastron meter to the ESP when running the reaper firmware, please also share it here :)

LeeNX, 2018-08-13:
Spent the weekend setting up my PoC for testing hardware and software. Ran into some stability problems with my NodeMCU WiFi, but I am not sure if this something in my toolchain or the newer/different NodeMCU device.  
  
 Wondering if when we include ModBus support, would we not just target the Eastron power meters or this would make the ESPEasy firmware possible too big? Basically just thinking about all the read registers of the Eastron meters - https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L50  
 
 Or would we just have a set of interfaces to registers and format of registers for reading?  
  
Have not yet even thought about writing into the ModBus devices ... Not sure if that is even worth thinking about?

s0170071, 2018-08-13:
related: https://github.com/letscontrolit/ESPEasy/issues/1466

TD-er, 2018-08-13:
I am not sure if I get what you meant to write, but maybe it will be clear when I explai a little about the current state of how ESPeasy does handle sensors.

- A plugin provides connection to a sensor device and is responsible for taking samples and collecting them in ESPeasy.
- A plugin presents up-to 4 float values as sensor values and optionally provides those samples to one or more controllers.
- A controller is responsible for sending values to another host in a specific format using a specific protocol (e.g. Domoticz HTTP/Domoticz MQTT/Thingspeak/etc.)

On top of that, there are some rules that can act on events or timers and perform actions like taking plugin values and sending them to controllers, etc.

There are sensors that provide a lot more sensor data than the 4 available value positions, so we still have to think about how to handle those. That's not yet a solved question.

LeeNX, 2018-08-13:
Thanks both @s0170071 and @TD-er ! You both send me down another rabbit hole!! ;-)  
  
Write options was more me thinking about getting something into the Eastron Meter, as there is an option to set things like secondary pulse option rate or other things, but you are quite right, that would be out of scope for ESPEasy, don't mind my rambling.   
  
Maybe I should move my question about supporting the large number of registers per ModBus device too #1466 ... It was just some brainstorming that left me with questions.  
  
I had done this before without the NodeMCU using python in OpenHAB, but every stormy season, my USB RS485 adapter would be hit by lightning and I would be without data again. Moving too NodeMCU and WiFi data collection using ESPEasy, I am hoping will remove the smoking electronics problem.  
  
As for offering any ideas on how to support more ModBus devices or attribs/registers, is a little outside of my wheelhouse.

TD-er, 2018-08-13:
You can ofcourse let a plugin send some parameters to the sensor device.
For example the SDS011 (dust sensor) and INA219 (Energy) do have such an option.
Also some plugins accept commands (sent from rules or other routes) which can be used to set some settings a bit more dynamically.

I guess changing settings on an Eastron meter doesn't have to occur that often, so either way could be acceptable.

LeeNX, 2018-08-13:
ModBus relay? ;-)  
  
Using ESPEasy platform as a bridge ... Crazy ideas

LeeNX, 2018-09-11:
Just received another MODBus/RS485 Meter ```DDS-238-2 ZN/S```, which I am thinking of hooking up to my eBike charger, so I know how much power/currency I am converting.  
  
Need to track down a manual and see how much might need to be done?  
  
Will start looking at this issue soon, thou I have so many things planned!!

LeeNX, 2018-09-11:
Link too ModBus Registers for DDS-238-1 ZN/S http://domoticx.com/modbus-kwh-meter-dds238-1-zn/  
  
Hope this is the same as DDS-238-2 ZN/S, soon too test, I hope.

IoTPlay, 2019-04-21:
Dis you make any progress?

TD-er, 2019-04-21:
For Modbus RTU, I made a separate interface, which is now used by the SenseAir plugin.
I'm also working on using that library to work with a MAX485 to address the DC power measurement units made by AccuEnergy.
So for the Modbus RTU part, there is some active development.

mttstt, 2019-04-23:
I made a plugin for Aurora Inverter Power DC. This is ok but there is a low memory problem, perhaps the plugin uses too many resources. To go I need delete some other plugin on a wemos 4M.
Probably the new Modbus RTU class can resolve my problem. But I am not able to implment it.

https://github.com/mttstt/remote-control-shutters/blob/master/ESP8266/Espeasy%20Plugin/_P111_Aurora.ino

TD-er, 2019-04-23:
@mttstt Maybe you can open a new issue about it.
The update of ModbusRTU is not yet merged (worked on it last night) so the patch to use the DE/RE pin is not yet included in the mega repository.
You can see the progress here: https://github.com/letscontrolit/ESPEasy/pull/2439  (and also get some inspiration)

One thing that's consuming a lot of memory is all the strings you're using. None of them are wrapped in `F()` macro tags, so they will be put in memory and not just read from flash.

pite81, 2019-09-17:
> I am not sure if I get what you meant to write, but maybe it will be clear when I explai a little about the current state of how ESPeasy does handle sensors.
>
> * A plugin provides connection to a [...]

I just try to work around the Eastron SDM630, managed to flash a firmware (ESP_Easy_mega-20190523_test_core_260_sdk222_alpha_ESP8266_4M.bin) but it keeps crashing, nothing connected yet. Even on soft reset, it struggles, only a cold start help it to recover. 
I have tried to upload a custom sketch with Arduino following this, https://www.letscontrolit.com/wiki/index.php/Tutorial_Arduino_Firmware_Upload. came up with an error. Same error in platformio. The goal would to disable the unwanted modules and plugins and use the ones I need to increase stability. Ideally, I just need an SDM630 and DS18, DHT22 for this purpose. I will measure the status of a heat pump, ie power consumption, the temperature of the hot water cylinder, temp and hum of the utility room. That is it. Some reason the older version ESP tend to solid and does not require a restart, the 2019 version crashes a lot. Any possibilities for that? Thanks for the brilliant firmware! You are a Genius!

TD-er, 2019-09-17:
Maybe you should have a look at the [Vagrant build option](https://github.com/letscontrolit/ESPEasy/issues/2594)
About the restarts.
The last few weeks/months we've made quite a lot of progress in reducing the number of restarts.
Most of them appeared to be related to WiFi and how it is handled.

I am not entirely sure how the SDM plugin does handle not connected modules.