letscontrolit/ESPEasy #1020: Deep-Sleep in PUYA-image not work

Micha-he, 2018-03-05:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Erase ESP-01 complete, than flash a PUYA_1024_VCC-image. Enable sleep-mode.

Does the problem persists after powering off and on? YES

### Expected behavior
ESP goes into deep-sleep after a few seconds.

### Actual behavior
(Deep-)Sleep mode is not activated. Web-interface is still accessible after minutes/hours.
But sleep-mode in the web-interface is checked.

### System configuration

Hardware:
ESP-01 with 1MB PUYA-flash (maybe it's only with this type)

Software or git version:
PUYA_1024_VCC-image from Release v2.0-20180228 & Release v2.0-20180227

### Associated forum post
[](https://www.letscontrolit.com/forum/viewtopic.php?p=25548#p25548)

Micha-he, 2018-03-07:
Many tests later....

If I powered the ESP-01 with two AA-bateries (~2.7V measured) instead of USB>3.3V-adapter, the same image/setup/ESP get into deepsleep. No idea why !?!?
I'll check the ESP-powersupply with an oscilloscope in the next days...

TD-er, 2018-03-07:
Maybe add some small (100nF) and bigger (470uF) capacitors to the 3V3 line?

Micha-he, 2018-03-07:
I'll look at that, but in my opinion there are both capacitors on the board, near the 8-pin connector.

TD-er, 2018-03-07:
Those are probably 10 - 100 nF at most?

Micha-he, 2018-03-07:
Yes, on a default USB-TTL UART maybe, but the second test-board is self-build (and there should be both capacitors added). And both boards have the same problem !?

s0170071, 2018-03-07:
@Micha-he : did you connect RST and D0 ?

TD-er, 2018-03-07:
Maybe a ground issue? When powered via batteries, the ground for power is floating.

Micha-he, 2018-03-07:
@s0170071 : Yes, RST is connected via magnetic (reed)-contact with GND, to reboot the system. This works flawlessly. GPIO0 is free. But what should that have to do, with the DeepSleep-event ?

s0170071, 2018-03-07:
The wake pin and RESET need to be connected to wake up from deep sleep. May not even be GPIO0
https://escapequotes.net/esp8266-wemos-d1-mini-pins-and-diagram/ labels that pin as 
"D0 4 IO16 WAKE 0 16" (upper right corner)

Micha-he, 2018-03-07:
The oszi shows nothing unusual. Capicitor on the CH340-converter is 1000μF. Should be enough.
Here are the two log, first faild one with USB->3.3V-power-supply, second with batteries is ok:
`304 : INIT : Booting version: v2.0-20180306
304 : INIT : Cold Boot
304 : FS : Mounting...
310 : FS : Mount successful, used 76304 bytes of 113201
366 : INIT : Free RAM:15480
367 : INIT : I2C
367 : INIT : SPI not enabled
1254 : WIFI : AP Mode disabled
1254 : WIFI : AP Mode ssid will be FK002_13 with address 192.168.4.1
1255 : WIFI : Connecting SBerg15 attempt #1
5272 : WIFI : Connected! IP: 192.168.99.111 (FK002_13)
5280 : WD : Uptime 0 ConnectFailures 0 FreeMem 13024
5280 : MQTT : Intentional reconnect
5341 : MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:1C:49:1B
5348 : Subscribed to: /FK002/#
35279 : WD : Uptime 1 ConnectFailures 0 FreeMem 12600
65279 : WD : Uptime 1 ConnectFailures 0 FreeMem 12600
96367 : WD : Uptime 2 ConnectFailures 0 FreeMem 12600
126366 : WD : Uptime 2 ConnectFailures 0 FreeMem 12432

297 : INIT : Booting version: v2.0-20180306
297 : INIT : Cold Boot
297 : FS : Mounting...
303 : FS : Mount successful, used 76304 bytes of 113201
359 : INIT : Free RAM:15480
359 : INIT : I2C
359 : INIT : SPI not enabled
1247 : WIFI : AP Mode disabled
1247 : WIFI : AP Mode ssid will be FK002_13 with address 192.168.4.1
1247 : WIFI : Connecting SBerg15 attempt #1
5264 : WIFI : Connected! IP: 192.168.99.111 (FK002_13)
5271 : WD : Uptime 0 ConnectFailures 0 FreeMem 13024
5272 : MQTT : Intentional reconnect
5327 : MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:1C:49:1B
5334 : Subscribed to: /FK002/#
5340 : SLEEP: Entering deep sleep in 30 seconds.`

uzi18, 2018-03-07:
cap is on 3V3 or on 5V side?

Micha-he, 2018-03-07:
The self-build board has also a 470μF after the voltage regulator and a 100nF near the ESP-01 connector.
Power supply shows no problems, but no deepSleep. 

@uzi18 : The cap is on 3V3.

TD-er, 2019-11-04:
This should no longer be an issue, so I will close this one.
Please let me know if it is still an issue.

Micha-he, 2019-11-05:
The issue still exists. See my reference above. The problem there, probably also applies here.

uzi18, 2019-11-05:
please try latest builds

Micha-he, 2019-11-05:
I've tested it today, with the release from yesterday.

Micha-he, 2020-03-29:
In my opinion, this issue only exists, because there is no pullup at GPIO-16 and the supply voltage is over a limit of ~3V.
I close this issue, because the other one still exists.

closed 2020-03-29 by Micha-he