letscontrolit/ESPEasy #1021: Webinterface: Missing menu-items

pwassink, 2018-03-05:
Hello,

This morning i tried the daily GIT version:	mega-20180305 on a Esp12F with 4 Mb, the "normal" image .
Still found that when logging in to the the esp it does not come up with the normal main screen, and the two menu options on the right of the menu-bar are missing completely. options missing : Notificatons / Tools.
When you press main on the menubar directly after logging in you get the udp-nodelist as it should and the missing items appear and will stay.

Might be a small issue, but seems wrong the way it works now ?

Was asked to open an issue on Git did that under the MEGA branch, seemed wrong, so here instead ..

TD-er, 2018-03-05:
Do you access the web interface from a mobile device?

pwassink, 2018-03-05:
No, desktop, even with different browsers give same results.

TD-er, 2018-03-05:
Can you post a screenshot?

pwassink, 2018-03-05:
Check,  Just after logging in (chromium browser gitversion one daily older mega-20180304 ! )
[image]
And after one enter you get this:
[image]
[image]

mikeluyten, 2018-03-05:
Can confirm, able to replicate on a 1MB esp8266 and a 1MB esp8285

TD-er, 2018-03-05:
You both do have a password set?

pwassink, 2018-03-05:
I have

mvdbro, 2018-03-05:
Is this thema same issue:
[image]

pwassink, 2018-03-05:
i dont think so, 

The two menu items on the right are missing on the first screen after a passwd-protected login, 
the one you showed has them both but on the wrong place (and through mobile-interface)

s0170071, 2018-03-05:
is this the mobile view or the desktop version of the page ? Mobile view is usually a mess...

mvdbro, 2018-03-05:
This is the mobile viewing that's now default since someone seems  to have added viewports. Control is a bit clumsy now on android at least

uzi18, 2018-03-05:
We know it, something is broken from time viewport added. Login page also is broken.

mvdbro, 2018-03-05:
Wouldn't it be better to revert the change until it actually works?

TD-er, 2018-03-05:
@doxikus was working on a proper change.
Maybe he can give an estimate on when it will be ready?

TD-er, 2018-03-06:
@doxikus created a quick fix, which I just merged.
In short, with a small screen (max width 960px) the menu items will be positioned vertical.
[image]

Please test on different devices.

s0170071, 2018-03-08:
@TD-er did you know that there is a web developer feature in firefox : " test with different screen sizes" ?

TD-er, 2018-03-08:
Yep, it also is a standard feature in chrome.
But that doesn't test content-length headers, which were causing issues on iOS devices.

pwassink, 2018-03-08:
On my esp07S and esp12* based devices the fault was fixed

closed 2018-03-15 by TD-er