letscontrolit/ESPEasy #1023: new plugin state: PLUGIN_PREPARE_READ

s0170071, 2018-03-05:
often when you read a sensor, often you need to communicate via I2C or TCP/IP to fetch the data. This is blocking, you wait for the communication to succeed or time out. 

If there was an additional state which is called prior to read, say 1s or  before, you could issue the request before (PLUGIN_PREPARE_READ) so that the answer is already there when the sensor is actually read. 

Example: I want to read something via TCP, e.g. my MODBUS inverter. I would issue the request (client.connect() + client.write()) on prepare and when I actually do read the sensor,  client.available() would already have the data I requested.

TD-er, 2018-03-05:
Some plugins do this by using the ten_per_second or once_per_second loop to fetch some packet, decode it and store the value as "last read".
Those are plugins where the reader has no influence on when the new reading will arrive. 

I am all in favor of splitting tasks over several loop-runs, compared to one do-it-all run.

Will this be a command, like "PLUGIN_ONCE_A_SECOND"?
Or a state for use per task?

I think keeping a state of a task (not in the plugin, but in a task to avoid problems running more instances of the same plugin) will make it more dynamic.
Then you can go through the entire state machine by yourself and only do little steps at a time only when data is ready.

uzi18, 2018-03-05:
@TD-er this will add more responsiveness, but will eat more RAM, as buffering/data analyse will be delayed in time.

TD-er, 2018-03-05:
Not if you try to re-schedule the sampling of the tasks.
When set at 30 sec interval, they do not have to be started at the same moment.

Using the software serial already buffers. If you can clear this buffer as soon as possible and decode the message, then the allocated buffer can be smaller.
The same for other plugins. Now buffers will remain allocated throughout the entire life of the plugin, but if you process them when needed, the data can be reduced. Often a data packet is a lot bigger than the actual values you want to get from it.

uzi18, 2018-03-05:
add callback in EspEasySofwareSerial if all needed data arrived?

TD-er, 2018-03-05:
How does the SoftwareSerial know when all data has arrived?

Often there is a stream of data in which you have to find the start of a packet, wait until the packet is complete (ask softwareSerial for buffer used) and then parse it.
The software serial has to be simple, since those routines will use the very limited iRAM. The periodical checking + analysis can be done in routines which are read from flash before execution.

s0170071, 2018-03-05:
Lets assume I execute something @ PLUGIN_ONCE_A_SECOND- can the plugin determine when it is due for reading ? If so, I could use this to issue my read command one second before a planned read. 
Only issue: If I schedule the plugin every second I may get race conditions...

uzi18, 2018-03-05:
My idea is to set "packet length" in SoftwareSerial. If packet is filled so we use callback in plugin to analyse data so less memory is used here.

TD-er, 2018-03-05:
@uzi18 That will only work when you know when the start is sent and where in the packet you are at reading.
If you can trigger receiving a packet, like with Modbus or M-bus, then that's a good solution for sure.
But not all plugins are so "developer-friendly".

@s0170071 each task already has its own "timer" set. These are defined in ESPEasy.ino.

The start state is "idle".
So when you call the 'periodical' functions in "idle state", you can just check if the timer has passed. If so, set the internal state to something like "prepare read" and send the requests.
On the next call, see you're at "prepare read" and set the state to "collect reading".
Only when the entire data packet is ready, set it to "new value present". If some timeout has been reached, or the data is invalid, either set to "prepare read" or "idle" and set the task timer again.

uzi18, 2018-03-05:
Yes but You can sync there and read until end of buffer or magic bytes found, and next callback will be there what You need :)

TD-er, 2018-03-06:
What happens if you set the callback trigger to let's say 16 bytes, that level has been reached and you get the call-back.
While you're reading the buffer gets bytes from the next message and thus it passes that level again while processing?

uzi18, 2018-03-06:
You read only 16 bytes,  how fast communication you want to support :)
Can process another pkt or drop next one.
It depend on what You want to do in callback.

Normaly what you do in such condition,  process all?

s0170071, 2018-03-06:
I will look into the timers later. For now I am monitoring the age of the measured values. 
```
if (ageOfValue == small) && (request already sent) {
  read data;
  send next request;
} else {
 send request;
 wait for answer & read data;
 send request;
}

```
This way if I have a tight poll interval, there is already a value present if the plugin is called. If the plugin interval is larger than timeout, I need to request a fresh value first. But in that case the plugin only blocks in large intervals ;-)

real code: 
 ```
    
   if (!Plugin_097_SMA->requestTimeout() && Plugin_097_SMA->requestSent){
       if (Plugin_097_SMA->getData())
           Plugin_097_SMA->sendRequest(); // send new request only if there was an answer first.
   } else {
       Plugin_097_SMA->sendRequest();
       while( !Plugin_097_SMA->requestTimeout() && Plugin_097_SMA->requestSent) 
          Plugin_097_SMA->getData();
       Plugin_097_SMA->sendRequest();
   }

```

s0170071, 2018-03-06:
@TD-er 

> Why end with a sendRequest?

Course you don't know when your next read is. Worst case: its within a few seconds.

> And why in the first part check the result of getData and not in the second part?

This first part is called when you call the plugin often. It requires there has just been a request. Whis has not timed out. And. there is getData in the second part too.

> And why no delay in the while loop? (may crash the ESP)

Which delay ? requestTimeout() returns true if the last request for data was too long ago.

You always start with the second  part because there has not been any request for data. After that, if you hurry, you end up in the first part.
What you don't see is that timeout() discards any answers if they originated from requests made too long ago. I admit its a brain twister and thats the exact reason why I asked for an pre-read event. To prepare reads by issuing requests in time.

TD-er, 2018-03-06:
About the delay part... If you're running a while loop like that, it will run as fast as it can. If the calls you make also do not call delay() somewhere, there is no call to background tasks (delay function will call yield())
If interval between background tasks is too long, the WiFi will loose its connection and also the watchdog timer will kick in => crash.

s0170071, 2018-03-06:
There is no while. Just if(...)
I'll PR to the playground later today.

s0170071, 2018-03-08:
The inverter does only allow MODBUS reads every 10 seconds. If I read more often, it becomes unresponsive after a while. So I actually have my communications state machine getting called by the 10persecond task call. 
What I noticed, however, is that 10persecond is called in irregular intervals that are more like half a second. The only other device active was the framed oled :-O Is it possible that it is pretty blocking sometimes ?

TD-er, 2018-03-08:
Oled scrolling is pretty blocking and also the web interface, DNS lookup, mqtt connect and ping when the other side is slow in responding.
The ping I will change into a non blocking version

closed 2019-10-11 by Grovkillen