letscontrolit/ESPEasy #1024: lowercase issues in events

uzi18, 2018-03-05:
git compare v2.0 and mega shows something like this:

```
diff --git a/src/_P067_HX711_Load_Cell.ino b/src/_P067_HX711_Load_Cell.ino
index 10af567..c1d2019 100644
--- a/src/_P067_HX711_Load_Cell.ino
+++ b/src/_P067_HX711_Load_Cell.ino
@@ -266,9 +266,8 @@ boolean Plugin_067(byte function, struct EventStruct *event, String& string)
 
     case PLUGIN_WRITE:
       {
-        String lowerString=string;
-        lowerString.toLowerCase();
-        String command = parseString(lowerString, 1);
+        string.toLowerCase();^M
+        String command = parseString(string, 1);^M
 
         if (command == F("tare"))
         {
diff --git a/src/_P070_NeoPixel_Clock.ino b/src/_P070_NeoPixel_Clock.ino
index cfb6233..b9bbb8e 100644
--- a/src/_P070_NeoPixel_Clock.ino
+++ b/src/_P070_NeoPixel_Clock.ino
@@ -144,12 +144,11 @@ boolean Plugin_070(byte function, struct EventStruct *event, String& string)
 
     case PLUGIN_WRITE:
       {
-        String lowerString=string;
-        lowerString.toLowerCase();
-        String command = parseString(lowerString, 1);
-        String param1 = parseString(lowerString, 2);
-        String param2 = parseString(lowerString, 3);
-        String param3 = parseString(lowerString, 4);
+        string.toLowerCase();
+        String command = parseString(string, 1);
+        String param1 = parseString(string, 2);
+        String param2 = parseString(string, 3);
+        String param3 = parseString(string, 4);
 
```

do we need to modify referenced strings?

TD-er, 2018-03-05:
That's one of the bugs I was fixing last week.
I don't get the diff here. It was changed from changing from modifying the referenced string to copy first and then tolower...
But your diff shows the opposite.

uzi18, 2018-03-05:
- is v2.0
+ is mega

TD-er, 2018-03-05:
Are you sure you're using the last checkout of the Mega branch?
I took those fixes from the Mega branch into v2.0.
And a quick check of the source of P070 here in both branches on Github, show the same (proper) code.

uzi18, 2018-03-05:
ok did not pulled actual v2.0 into account ;)

closed 2018-03-05 by uzi18