letscontrolit/ESPEasy #1029: ESP32 Reboots every 18 minutes with framed oled enabled

Budman1758, 2018-03-07:
I have an ESP32 running the latest version (20100) and I have encountered this bug.  The unit has a Dallas temp sensor, a standard OLED and a Framed OLED.  All of this works just as its supposed to except that when the framed OLED is enabled the unit reboots every 18 minutes. You can set your watch to it because its that consistent. I set a syslog server up to watch it overnight and it rebooted EVERY 18 minutes. No if, ands or buts.
This issue goes away when the Framed OLED is disabled. (enabled checkbox in the plugin).  As of now the uptime is over 180 minutes.

Let me know what other info you may need to help.

Here is the serial output when it reboots.....

WD   : Uptime 17 ConnectFailures 0 FreeMem 106448
UDP  : Send Sysinfo message
UDP  : 5C:CF:7F:ED:78:4E,192.168.1.34,5
UDP  : 5C:CF:7F:ED:79:14,192.168.1.30,1
UDP  : 5C:CF:7F:ED:79:77,192.168.1.32,3
DS   : Temperature: 23.31 (28-ff-2-c5-83-16-3-51)
EVENT: dallas1#temp=73.96
EVENT: dallas1#temp=73.96 Processing time:130 milliSeconds
EVENT: Clock#Time=Wed,09:11
EVENT: Clock#Time=Wed,09:11 Processing time:134 milliSeconds
UDP  : 60:01:94:00:FE:82,192.168.1.11,11
UDP  : 5C:CF:7F:56:15:13,192.168.1.184,21
UDP  : 5C:CF:7F:ED:79:D2,192.168.1.33,4
UDP  : 5C:CF:7F:ED:78:4E,192.168.1.34,5
Memtrace
        0: lowest: 104468  PluginCall_s (20)-> 106252 PluginCall_s (3)-> 106252 PluginCall_s (32)-> 106252 PluginCall_s (20)-> 106252 PluginCall_s (3)-> 106252 PluginCall_s (32)-> 106252 PluginCall_s (20)-> 106252 PluginCall_s (3)-> 106252 PluginCall_s (32)-> 106252 PluginCall_s (20)-> 106252 PluginCall_s (3)-> 106252 PluginCall_s (32)-> 106252 PluginCall_s (20)-> 106252 PluginCall_s (3)-> 106252 PluginCall_s (32)-> 104468
1: lowest: 65535
2: lowest: 65535
WD   : Uptime 18 ConnectFailures 0 FreeMem 106448
UDP  : Send Sysinfo message
UDP  : 5C:CF:7F:ED:79:14,192.168.1.30,1
UDP  : 5C:CF:7F:ED:79:77,192.168.1.32,3
UDP  : 60:01:94:00:FE:82,192.168.1.11,11
UDP  : 5C:CF:7F:56:15:13,192.168.1.184,21
UDP  : 5C:CF:7F:ED:79:D2,192.168.1.33,4
UDP  : 5C:CF:7F:ED:78:4E,192.168.1.34,5
Memtrace
        0: lowest: 104304  parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105848 parseTemplate3-> 105768 parseTemplate-> 105816 parseTemplate3-> 105756 Calculate-> 104304
1: lowest: 65535
2: lowest: 65535
WD   : Uptime 18 ConnectFailures 0 FreeMem 106444
UDP  : Send Sysinfo message
UDP  : 5C:CF:7F:ED:79:14,192.168.1.30,1
UDP  : 5C:CF:7F:ED:79:77,192.168.1.32,3
DS   : Temperature: 23.25 (28-ff-2-c5-83-16-3-51)
EVENT: dallas1#temp=73.85
EVENT: dallas1#temp=73.85 Processing time:128 milliSeconds

Stack smashing protect failure!

abort() was called at PC 0x401121ab on core 1

Backtrace: 0x400883f8:0x3ffd64d0 0x400884f7:0x3ffd64f0 0x401121ab:0x3ffd6510 0x401061b9:0x3ffd6530 0x400f46f7:0x3ffd6580 0x400f99d2:0x3ffd6640 0x400f9a40:0x3ffd6660 0x400fd24f:0x3ffd6690 0x40150b70:0x3ffd66c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
▒U

INIT : Booting version: (custom)
INIT : Cold Boot
FS   : Mounting...
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK
INIT : Free RAM:169568
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy 32_0 with address 192.168.4.1
WIFI : Connecting xxxxxxxxxx attempt #1
NTP  : NTP send to 38.126.113.10
NTP  : NTP replied: 48 mSec
Current Time Zone:  DST time start: 2018-03-11 02:00:00 offset: -420 minSTD time start: 2018-11-04 02:00:00 offset: -480 min
WIFI : Connected! IP: 192.168.1.115 (ESP_Easy-32_0)
OTA  : Arduino OTA enabled on port 8266
UDP  : Send Sysinfo message
NTP  : NTP send to 38.126.113.10
NTP  : NTP replied: 58 mSec
EVENT: System#Boot
EVENT: System#Boot Processing time:132 milliSeconds
Memtrace
        0: lowest: 65535
1: lowest: 65535
2: lowest: 65535
WD   : Uptime 0 ConnectFailures 0 FreeMem 106448
UDP  : Send Sysinfo message
DS   : Temperature: 23.31 (28-ff-2-c5-83-16-3-51)
EVENT: dallas1#temp=73.96
EVENT: dallas1#temp=73.96 Processing time:129 milliSeconds
UDP  : 60:01:94:00:FE:82,192.168.1.11,11
EVENT: Clock#Time=Wed,09:12
EVENT: Clock#Time=Wed,09:12 Processing time:140 milliSeconds
UDP  : 5C:CF:7F:56:15:13,192.168.1.184,21
UDP  : 5C:CF:7F:ED:79:D2,192.168.1.33,4
UDP  : 5C:CF:7F:ED:78:4E,192.168.1.34,5

TD-er, 2018-03-07:
Is it always with this error:

> Stack smashing protect failure!

And when you set the update interval of the display different, does the 18 minutes change?
And what about when changing the scroll speed?

Budman1758, 2018-03-07:
I will try that.  One at a time or both at once?

TD-er, 2018-03-07:
One at a time, since either one or both may affect it and not sure how.

Budman1758, 2018-03-07:
OK.  Will try interval first

Budman1758, 2018-03-07:
Methinks your onto something.  I doubled the display interval from 10 to 20 seconds.  Now the unit reboots at 36 minutes uptime.  Exact same error message.

TD-er, 2018-03-07:
Do you use the OLED library included with ESPeasy ?

Budman1758, 2018-03-07:
I'm using the bin file from here.
https://www.letscontrolit.com/wiki/index.php/ESPEasy32#Firmware

TD-er, 2018-03-07:
It states to use the latest sources + lib from March 4.
These have the latest patches from me in the OLED library. A lot of these patches were about not freeing allocated memory.
So maybe there is something with not releasing strings, or in the I2C communication.
Perhaps "Wire" is different for the ESP32?
If scroll-speed set to "instant" will increase the crash interval period, then it may be something with I2C communications.
It is also possible that implementation uses Brzo I2C library.
I did not change anything with the OLED lib for use with that I2C library, but I would not be surprised that combination will have some memory leaks. There were lots in the OLED library.

Budman1758, 2018-03-07:
I will try 10 second interval and instant scroll and report back,

TD-er, 2018-03-07:
I think I can stay awake for another 18 minutes ;)

Budman1758, 2018-03-07:
18 minutes.  Just like clockwork.  Seems the delay interval is the deciding factor.  Scroll speed seems to not affect it.

TD-er, 2018-03-07:
OK, then it is up to @mvdbro to give some queues on used libraries :)

Budman1758, 2018-03-07:
This is a pretty time consistent bug.  I changed the scroll interval to 5 seconds and the uptime before it rebooted was exactly 9 minutes.

Budman1758, 2018-03-08:
Disabled the Framed OLED.
Uptime now 860 minutes

Budman1758, 2018-03-30:
With the newest ESP32 bin file made available on 3/30 this bug is still the same.

@s0170071 I don't know if your #1030 was targeted at this issue or not.  If it's in the code now it seems to not affect this issue. :-(

TD-er, 2018-04-19:
I have the ESP32 running for close to 18 hours now, with the OLED framed active.
It is logging the free memory to Domoticz over MQTT.
I guess there is a memory leak somewhere.... 
[image]
Now the question, where?

s0170071, 2018-04-19:
What's the granularity of the leak?

TD-er, 2018-04-19:
About 1.5k in 6 hours.
Animated scroll on the OLED, which is refreshed every 5 seconds.

s0170071, 2018-04-19:
I mean how many bytes does it lose at a time?

closed 2019-10-11 by Grovkillen