letscontrolit/ESPEasy #1031: ##BUG## 0.91" Oled screen not working randomly (Heltec HTIT-W8266)

Nico34160, 2018-03-08:
Hello,

I have bought a ESP8266 with an embeded 0.91"  Oled screen (128x32), after installing ESPEasy in MEGA version (to have OLED 128x32 choice available) a bug is present, the screen work for few seconds then shutdown.
At this time no I2C device is found (with I2C SCAN tool).

I have uploaded a test code provided by the manufacturer (Arduino code) the screen work fine and does not shutdown anymore.

Maybe you're aware of this bug and a solution is available but i haven't find any topics on this issue.

Could you help me ?

### Steps to reproduce
Upload ESPEasy MEGA onto HTIT-w8266 board (Heltec wifi kit8).
Create OLED device SD1306.
Tick "enabled"
Choose 128x32 for the screen size
Click submit.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Yes
### Expected behavior
Tell us what should happen?
The screen would  does not shutdown in normal case.

### Actual behavior
Tell us what happens instead?
the screen shutdown after few seconds

### System configuration

Hardware:
Heltec HTIT-W8266 / WIFI KIT 8

Software or git version:
ESP_Easy_mega-20180307_normal_ESP8266_4096.bin

Below you will find the Github from the manufacturer for this module, the link have some library available for Arduino maybe this can help you to find the problem.

Regards.

TD-er, 2018-03-08:
Do you have a link to the sketch that works?

TD-er, 2018-03-08:
Never mind, saw the link.,.

Nico34160, 2018-03-08:
the link for the working script (arduino).

[https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp8266/libraries/OLED/examples/OLED_091_128x32_SSD1306/OLED_091_128x32_SSD1306.ino](url)

TD-er, 2018-03-08:
Have you tried setting the display to "medium" contrast?
That can be done in the web interface, or by sending a command: 
http://192.168.1.75/control?cmd=OLEDFRAMEDCMD,med
Ofcourse use the IP of your own module.
You can also try "low", "med", "high", "on" or "off"

TD-er, 2018-03-08:
Oh and by the way, please also test the "OLED Framed" plugin

Nico34160, 2018-03-08:
Oled framed display nothing on the screen.
I will try with adjusting contrast on the OLED.
I will keep you informed.

Nico34160, 2018-03-09:
Ok i have made the test, i haven't contrast settings on OLED plugin (not framed OLED), just "on", "off", "clear".
if the screen display anything the command (on, off, clear) works fine but when the screen shutdown the command doesn't work anymore.
Strange things is the  lost of the I2C module, when the problem occur, the I2C tool display "No devices found" instead of : 0x3c PCF8547A OLED.
If the OLED screen is disabled the I2C module never disappear.
Another strange thing, the I2C module is lost when the screen is refreshed.

TD-er, 2018-03-09:
Did you test with the OLED Framed to switch the contrast level, to see if
you get something displayed?
N.B. do not use both OLED plugins at the same time on the same I2C address.
So either remove the other, set to other I2C address or disable the other
plugin.

---
Verzonden vanaf laptop

Nico34160, 2018-03-09:
test with OLED Framed : Nothing displayed.
adjusting contrast doesn't work.
I2C devices are missing (no devices found).
If i disabled OLED device, the I2C devices stay detected all the time.

TD-er, 2018-03-09:
That sounds like the OLED device is crashing and blocking all I2C communications.
Do you have a link where these are being sold?

Nico34160, 2018-03-09:
Why OLED crash using ESPEasy firmware and not when using test code arduino (test code use also Wire.h library so the display is driven through the I2C channel...)

The link for bought the module: 
[https://www.amazon.com/ESP8266-development-Arduino-NodeMcu-0-91inch/dp/B074373KTY/ref=sr_1_6?ie=UTF8&qid=1520603485&sr=8-6&keywords=esp8266+oled](url)

TD-er, 2018-03-09:
The OLED is connected to GPIO-2 (SDA) and GPIO-14 (SCL) and a separate reset (GPIO-4)
The current implementation of the OLED plugins doesn't use any reset and maybe something is being done with GPIO-4 in ESP-easy?
I assume you have nothing connected to that pin?
Maybe you could set the internal pull-up resistors for that pin in the configuration page?

Nico34160, 2018-03-09:
Nothing connected on the module.
how i can set an internal pull-up resistor in ESPEasy ?

Nico34160, 2018-03-09:
pay attention to the schematic provided by the seller, the only way to have anything on the screen is to set I2C (in Hardware tab) has default ESPEasy config (GPIO<->SDA = GPIO-4(D2) and GPIO<->SCL = GPIO-5(D1).
I have tested with the GPIO02 and GPIO14 but in this case nothing on screen at all time and I2C devices are not detect even if OLED is disabled.

Nico34160, 2018-03-09:
In the test code (working fine) i have see this :  
`// This example just provide basic function test;
// For more informations, please vist www.heltec.cn or mail to support@heltec.cn

#include <Wire.h>
#include "OLED.h"

//WIFI_Kit_8's OLED connection:
//SDA -- GPIO4 -- D2
//SCL -- GPIO5 -- D1
//RST -- GPIO16 -- D0

#define RST_OLED 16
OLED display(4, 5);

// If you bought WIFI Kit 8 before 2017-8-20, you may try this initial
//#define RST_OLED D2
//OLED display(SDA, SCL);

void setup() {
  pinMode(RST_OLED, OUTPUT);
  digitalWrite(RST_OLED, LOW);   // turn D2 low to reset OLED
  delay(50);
  digitalWrite(RST_OLED, HIGH);    // while OLED is running, must set D2 in high`

So the good settings for I2C is : GPIO<->SDA = GPIO-4(D2) and GPIO<->SCL = GPIO-5(D1).

If you look just below "void setup" reset pin is set to low and after 50ms set to high....
Maybe this informations can help you

Nico34160, 2018-03-09:
Ok, you're on the good way with the internal pull-up.
I have set GPIO16 to ouput high and now the screen stay on :)

TD-er, 2018-03-09:
Hmm, GPIO-16 is a special one, at least in ESPeasy.

This may prevent the unit to go into deepsleep and I wonder if it will restart properly.

Nico34160, 2018-03-09:
Thank you for your help :)

TD-er, 2018-03-09:
Does the Framed plugin work, now the reset is managed?
That one may allow for larger fonts, when using 1 line per screen.
And also the animation may give a quick indication on its activity. Also it will show the WiFi connection strength.

Nico34160, 2018-03-10:
the screen doesn't work correctly with OLED Framed, some lines are missing or bad drawed but the screen stay "ON" with GPIO-16 = Output High

TD-er, 2018-03-10:
The 32 pixel height screens only show the odd (or even) lines.
Still have to fix that.

Nico34160, 2018-03-11:
If you need a beta tester for the OLED Framed plugin on 128x32 screen I can made tests.

Nico34160, 2018-03-13:
After few days, the OLED is working fine but the ESP8266 reboot every +/- 2 hours, it's not critical but i think this is due to GPIO16 settings and cabling onto this module.

TD-er, 2018-03-13:
Is it a constant interval?
If semi random, please try another USB cable and/or power supply.

tentious, 2018-03-21:
Would also like to get this guy going. I have this exact model and would like to feed some data to the 128x32 OLED screen. Let me know if you get a beta working, will be happy to test.

GerdW, 2019-01-22:
same problem here

TD-er, 2019-01-22:
Is it this board: [WIFI ESP8266 0.91 Inch Blue OLED Display WIFI Kit 32 IOT Development Board for Arduino with a heat sink](https://www.aliexpress.com/item/WIFI-ESP8266-0-91-Inch-Blue-OLED-Display-WIFI-Kit-32-IOT-Development-Board-for-Arduino/32837499036.html?spm=2114.10010108.1000013.1.332f455dY6p2tf&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.90158.0&scm_id=1007.13339.90158.0&scm-url=1007.13339.90158.0&pvid=d54a5a57-6779-4980-af18-5955716d692c) ?
If so, I wonder why they supply it with a heatsink.

GerdW, 2019-01-22:
looks like the same board. maybe the heat-sink is for the charging chip on the backside (it has a charging circuit)
the display remains dark with easyesp

sebr82, 2019-03-11:
Still somebody working on this issue?
Sample code from Heltec works flawless, but no luck with espeasy.
Schematic is available too: https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/SchematicDiagram/WIFI_Kit_8_Schematic_diagram.PDF
Plus the info they give about this module: "In order to reduce the pin number, we use a hardware RESET system. Before init OLED device, delay more than 10ms is must needed."

TD-er, 2019-03-11:
@sebr82 If I understand your reply correctly, it means you have to wait to init the device for a while after boot?

So if you enable the plugin after boot, it is working?

sebr82, 2019-03-11:
No, unfortunately not. That was just the information from Heltec regarding the OLED on this module. Situation is like in the initial post: if OLED plugin is enabled, it is not working plus also nothing on I2C detected anymore. 
The issue is likely to be the reset line. Any chance to get this sorted out?

TD-er, 2019-03-11:
GPIO-16 does not have a pull-up resistor like the other pins have.
You could try to add a switch plugin to that pin and set it to output "1".

The thing is, I don't have this board so I cannot test it. I can try to add a checkbox to pull it high, but it may be a rather iterative process to make some changes, make a build, let someone test it, try to get the answers without knowing exactly what's wrong, etc.

Also it is a 128x32 OLED, which is not really very well supported by the OLED framed plugin at this moment.

TD-er, 2019-03-12:
@rsoldie It looks like this particular module is keeping the OLED display in reset state.
So unless we release  this reset state, it will not work.

rsoldie, 2019-03-12:
danke für die schnelle antwort, aber wie kann ich das ändern?

TD-er, 2019-03-12:
It looks like we have to change something in ESPeasy itself, like adding another pin definition for the reset in the plugin.

But apart from that there also has to be made a few more changes in the plugin to support the 128x32 display.

rsoldie, 2019-03-12:
ich kann kein neuen code aufspielen. das gerät wird erkannt aber der upload wird mit fehler abgebrochen.

rsoldie, 2019-03-12:
die led unter BAT blinkt mit hoher frequenz ohne ende.

TD-er, 2019-03-12:
OK, I re-read the whole thread and before I do anything for it, there has to be something really clear here.
What are the pins used by this board?
I see a drawing which is using totally different pins compared to the PDF and also the linked source code that appears to be working.
Also it looks like there are several versions of this board.
And it seems like the board also supports a battery, so there should be some deepsleep wake functionality, which is using GPIO-16 to wake. But that one appears to be connected to the RST of the OLED.

So in short:
- What pin is the RST connected to?
- What version of the board is at use => also related to the pins
- Is the board capable of wake up after deepsleep? => then RST and D0/GPIO-16 are connected
- What pins are used for SDA/SDC? Some links suggest the normal GPIO-4 and -5, but there is also a schematic which suggests GPIO-2 and -14.
- What code does work?

rsoldie, 2019-03-12:
ich möchte nur einen wifi-test auf das board senden. die pinbelegung ist unbekannt. die bezeichnung der pins ist minimal abweichend von der abbildung oben.
[image]
ich hoffe das bild ist sichtbar.

TD-er, 2019-03-12:
Is there maybe something written at the bottom of the PCB which may relate to the year/week of production? Or some version number of the PCB?

rsoldie, 2019-03-12:
[image]
ich kann nichts finden, aber sieh selbst

rsoldie, 2019-03-13:
na schön, da das hier doch nicht sooo fruchtbar ist hab ich mir ein neues modul bestellt. neuinstallation von git und co haben auch nichts gebracht. danke für eure mühe.
mfg r.

sebr82, 2019-03-14:
> * What pin is the RST connected to?
>
> * What version of the board is at use => also related to the pins
>
> * Is the board capable of wake up after deepsleep? => then RST and D0/GPIO-16 are connected
>
> * [...]

There are indeed different versions of this board, most likely clones, out there but also wrong schematics. From what I know I have a legit Heltec board. The pinout/connections is exactly as shown here: http://stefanfrings.de/esp8266/index.html (Version A)
The following code (1st post) is running flawless on the board: https://www.mikrocontroller.net/topic/438027

TD-er, 2019-03-14:
@rsoldie Based on the images shown [here](http://stefanfrings.de/esp8266/index.html) it looks like you have a "version A".

[image]

Version A:
- GPIO-5 => SCL
- GPIO-4 => SDA
- GPIO-16 => reset display (low = reset ????)

version B:
- GPIO-14 => SCL
- GPIO-2 => SDA
- GPIO-4 => Reset display  (low = reset)

It think it should be possible to simply test with either GPIO-16 pulled high, or GPIO-4 pulled high to see if the display can be found via I2C.
If so, then it is no longer in reset.

sebr82, 2019-03-14:
> It think it should be possible to simply test with either GPIO-16 pulled high, or GPIO-4 pulled high to see if the display can be found via I2C.
> If so, then it is no longer in reset.

It is. Just connected a pullup resistor between GPIO-16 and 3.3V and the display works reliable now.

Would be great if the sequence from the code above could make it into espeasy for display init.

rsoldie, 2019-03-15:
ich danke euch sehr. inzwischen habe ich ein neues board da ich für das alte board immer neue fehlermeldungen bekam und der frust mich von meinem vorhaben abhält konstruktiv zu sein.
ich hoffe der chat hilft mir die gemachten fehler, welche auch immer das waren, zu vermeiden.
mit freundlichen grüßen
rs

RallyHagge, 2019-05-22:
I have the "A" version of the Heltec Wifi Kit 8 (AKA: HTIT-W8266)

I have the display working if I'll set GPIO-16 to 1 at boot, but it won't display anything until I click "Submit" in Devices -> Edit (Display - OLED SSD1306). If I reset or power off the device I have to go into the HTTP interface and click the Submit button again.

Added a rule for the GPIO-16 pin:

On System#Boot do
  gpio,16,1
endon

RallyHagge, 2019-06-03:
> It is. Just connected a pullup resistor between GPIO-16 and 3.3V and the display works reliable now.
>
> Would be great if the sequence from the code above could make it into espeasy for display init.

I wasn't able to get it working reliably by adding the pullup resistor. After being left w/o power for a minute the display was scrambled upon next boot.

However, adding a connection or a 1K resistor between RESET and GPIO-16 did the trick and hasn't failed me since.

[image]

Roos-AID, 2019-09-26:
I can confirm that I experienced the same issue and with the resistor it is OK.

WayneManion, 2019-10-31:
I had the same problem using this board with the Tasmota firmware. A ~1K resistor between RST and GPIO16 appears to have solved the problem for me. Thanks RallyHagge.

polanaut, 2019-11-17:
@WayneManion  @sebr82 Wondering if, after adding resistor, I2C SCAN tool can see the display? 
Scan on my board picks up 0x3C at times, but very inconsistent.

snefrika, 2020-08-29:
Trick with the resistor (200Ohm in my case) fixed the same issue I had. THX.

snefrika, 2020-08-30:
The 200Ohm resistor  improved but did not fix it :-(. The display still gets black after some time, while the rest of the device is still working.

Leyoupi, 2020-10-30:
Hi, this command "display.flipScreenVertically(); " call more than once gets my screen black.. => only call it in setup function

TD-er, 2020-10-30:
As far as I can see, it is only called in the init function.

tonhuisman, 2023-09-02:
This seems to be hardware specific, so I don't expect a 'fix' in ESPEasy, AFAICS this can be closed.

closed 2023-09-02 by TD-er