letscontrolit/ESPEasy #1038: Update Data and time using MQTT or HTTP. MQTT broker/ Controller (Pi) will publish the date and time with regular interval and all ESP will subscribe the same. Broker / Controller will interact with NTP server / RTC module. ESP not required to expose to internet. Internet / UDI will ensure the uptime for ESP network. Is this possible? I am not technical person.

sujitrp, 2018-03-10:
- **NOTE:

 This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

TD-er, 2018-03-10:
I moved the topic into the question field :)

About your question, you can run a NTP server on a Raspberry pi, to which all ESP modules will connect.
On the pi, it is probably something like `sudo apt-get install ntp`

Grovkillen, 2018-03-10:
I've done the same on a windows machine

sujitrp, 2018-03-10:
How to update Date and Time on ESP Easy powered 8266 ?  Is any MQTT subscribe topic available to update ESP date and time ? 
My Pi is perfectly updating NTP on pi boot . Using Python program and MQTT paho with Corn Job Pi can Publish the Pi system time (NTP) time .  During Boot /or using rule engine  ESP can subscribe this MQTT feed.  I  think this will doubly ensure that ESP clock rule.

TD-er, 2018-03-10:
Does it work when you enter the IP address of your raspberry Pi as host for the NTP service?
For this you have to install the NTP-daemon service on your Pi.
Update time via MQTT is not a good idea, since there is no guarantee when you will receive the message.

micropet, 2018-03-10:
I think the accuracy should be sufficient.

TD-er, 2018-03-10:
But why introduce some completely new protocol not supported by anyone, while we have NTP?
You can run an NTP service on any Linux based host and it's just a matter of one install command and using that host as NTP source.

Grovkillen, 2018-03-10:
Totally agree. You can have a local NTP server alongside your MQTT broker and the ESP units themselves will not be exposed to the internet.

papperone, 2018-03-10:
+1 
There are plenty of solutions to use external NTP or a local one installed on your netowrk.
I don't see any reason to add useless code to manually setup date/time

sujitrp, 2018-03-10:
Yes NTP-daemon service . This sets the time w r t own time zone on PI. To update Local time on ESP I have two option within intranet (Within Wifi zone) . 1.ESP can Pull time @ Boot . Here I need Pi to host the Local NTP server.  ESP Easy can read the NTP time @ Boot or @ Rule execution or intermediate Hourly check   
2. Make Pi Publish the Time over MQTT every second with QOS 0. @ boot subscribe this topic . @ rule Subscribe to this topic .  MQTT will locally set  or correct ESP  time. MQTT topic will make Locally made  global NTP server.
Whichever device want time it can subscribe to this topic .
No Need to expose your vulnerable IOT device to internet to get the time

mattlward, 2018-03-13:
I would have to agree about not using MQTT... there can be a time delay introduced an the MQTT import task is hard on the CPU.  Trust me, I was running 2 in an unsupported manner and have cut down to 1 task and none on a few units.  The performance just gets better.  NTP is ideal for the task at hand and likely more reliable than a broker.

sujitrp, 2018-03-14:
Phasing out the option of using MQTT for Time update . 
 Now Local NTP option.
 1. Make your Pi / Local Window or Linux machine as Local NTP server  with fix IP.
2. Use the same on your all ESP for clock update . 
Can I use Local IP (Local NTP server) instead of  or along with NTP flag & NTP Hostname ?

TD-er, 2018-03-14:
Sometimes a router also has a built-in NTP daemon running.
For example my Fritzbox has one running.

You can also use a DNS name to indicate a NTP server.
So use either an IP-address or hostname that will resolve to the machine running NTP daemon/service.
It doesn't matter if that machine is somewhere in your own network or on the internet.

Grovkillen, 2018-03-14:
Closing time?

sujitrp, 2018-03-14:
My Objective is run the Clock based ESP Rule without  or with internet .  Still my objective is not achieved. 
Can run my ESPeasy network with no Internet connection ?

TD-er, 2018-03-14:
Yes you can, as long as you can reach the NTP service in your local network.
The NTP daemon running on Linux should use the internal clock and only try to sync every now and then (when possible) with an external server.
If the NTP daemon cannot reach any outside NTP server, it should continue to work.

sujitrp, 2018-03-14:
Just updating My Local machine / pi  IP address in ESP Easy  NTP server  will be sufficient ?

TD-er, 2018-03-14:
Yes, as long as you're also running a NTP daemon (NTP server) on that computer.
For completeness, you may want to have a look at [this blog](http://raspberrypi.tomasgreno.cz/ntp-client-and-server.html)

sujitrp, 2018-03-14:
On Pi I am ready with this setup .  This setup is working perfectly for last 4 months on pi . I wanted to set up RTC for my pi but as alternative I found this blog. And post configuring now No need for RTC module. My Pi runs 24 * 7  with DIY battery backup. 
Today I will try this NTP time update by local NTP server on ESP Easy .

sujitrp, 2018-03-14:
It is working great ... thanks Gijs ,mattlward,  Grovkillen & Papperone

closed 2018-03-14 by sujitrp