letscontrolit/ESPEasy #1039: Topmost SSID are covered on mobile Browser

Micha-he, 2018-03-10:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Open webinterface for main-setup with mobile browser. Here 'VIA Browser' is used.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
YES

### Expected behavior
See all SSID's.

### Actual behavior
Only the lowest SSID are complete shown.

### System configuration

Hardware:
ESP-01, Galaxy S5 with VIA-Browser

Software or git version:
ESPEasy_v2.0-20180309

[image]

TD-er, 2018-03-11:
Could you please test again?

Micha-he, 2018-03-11:
It's the same, with the firmware from today :pensive:

TD-er, 2018-03-11:
Could you test on tomorrow's build?
I just merged a fix for the access point selection screen with > 960 pixels width.

Micha-he, 2018-03-12:
Now it's top !

[image]

closed 2018-03-12 by Micha-he