letscontrolit/ESPEasy #1042: Please add WaveShare 2.9 inch e-paper support

micropet, 2018-03-10:
E-paper displays are getting more and more popular for battery powered applications. WaveShare e=paper displays are nice and almost cheap. It would be fantastic to add WaveShare e-paper displays support in your excellent ESPEasy Project

Please add 2.9inch 296x128 pixels module to your list.
http://www.waveshare.com/w/upload/e/e6/2.9inch_e-Paper_Datasheet.pdf

TD-er, 2018-03-10:
I've already got one here for months. So there will be support for it, not sure when I've got time to do it.
These displays have a SPI interface and there is currently no SPI device supported in ESPeasy.
Not sure why.

micropet, 2018-03-10:
Great. Is not in a hurry.
Maybe the SPI support can be used also for other projects.

Grovkillen, 2018-03-10:
I want to order one for the future. Just a check, is this the one we're talking about?

2.9inch E-Ink display module,2.9'' e-paper,296x128,Three Display color: red,black,white, SPI interface,No backlight, wide angle,
 http://s.aliexpress.com/zIn2mERZ?fromSns=Copy to Clipboard

TD-er, 2018-03-10:
You probably don't want the one with red pixels.
I've got one and it appears to be the really slow one (8 second refresh)
Some of the b/w support even partial repaint, which will allow for sub-second refresh rates. (and less memory use)

Grovkillen, 2018-03-10:
Thanks @TD-er for the heads up. This one is even cheaper. If you give me a thumbs up I'll order one. 

2.9inch e-Paper Module.296x128,2.9''E-Ink display,SPI interface,For Raspberry Pi etc,Display color: black,white, partial refresh
 http://s.aliexpress.com/Afqqumu2?fromSns=Copy to Clipboard

TD-er, 2018-03-10:
Yep and on that same page there is a nice table with all the modules and refresh rates.

Don't get them too large, since each pixel needs a bit for frame buffer.
Or we should do more on the ESP32 ;)

micropet, 2018-03-10:
@Grovkillen 
Yes, last, without red, is right.

I also hope that ESPEasy will run on an ESP32.
The 8266 simply has too few connections.

Budman1758, 2018-03-10:
@micropet   
ESPEasy is running on an ESP32 right now.  Some stuff works, some does not but its definitely working.
https://www.letscontrolit.com/wiki/index.php/ESPEasy32#Firmware

micropet, 2018-03-10:
I know the page. I tried to compile the version myself, but had only error messages.
I do not want to use the finished .bin file.
I also believe that the version for the real use is not yet useful.

Budman1758, 2018-03-10:
@micropet  "I do not want to use the finished .bin file."

What have you got against the finished bin file???
If you have not used it yet how can you say it is not useful???
If you read the page you would know where the source files are.

micropet, 2018-03-11:
In a .bin file I can not change anything.
I want to compile the source.

I have read the manual several times and have taken all the steps.
With Arduino or PlatformIO I always get errors because of software serial and some more.

With not useful I mean that probably not all plugins (especially CO2 and dust) work.
Is there a github where I can find "ALL" source files?

Greetings Peter

mvdbro, 2018-03-11:
@TD: We actually use SPI on the SD-Card in ESP Easy, so this should be no issue to get it working for other devices. Although running multiple devices on the same bus could be an issue if CS is not implemented properly on all connected devices.

Misiu, 2020-12-29:
@TD-er any updates on the e-paper support?
1.54 display (https://www.banggood.com/1_54-Inch-E-ink-Screen-Display-e-Paper-Module-Support-Partial-Refresh-For-Raspberry-Pi-p-1365280.html?cur_warehouse=CN) is quite cheap and it would be very useful to display sensors values.

tonhuisman, 2020-12-29:
There already is support available for the IL3897 eInk boards for > 6 months. It's not easy to tell if the one you linked to is using that chip, but current release has it included in the `display` build.

Misiu, 2020-12-29:
@tonhuisman I'll order one for testing and we'll see :)

tonhuisman, 2020-12-29:
From experience I can say that all these small (< 2") displays ae quite hard to read when viewed from a bit of a distance (> 1 meter). I'd suggest to invest a little more and get either a 2.13", 2.7" or 2.9" display, they are still max. $20,- from Aliexpress or Banggood (I can order them for around €20,- locally, excluding shipping of ~€6,-, so it might be cheaper, but a bit more risky and slower, to order from Ali 😄)

melkati, 2021-03-04:
Is there any documentation about connections and a working example to test?
I stent some hour trying to make it work but can't.

tonhuisman, 2021-03-04:
Connecting a device via SPI isn't that hard, but it's not documented explicitly on either the Wiki or the ReadTheDocs pages, AFAICS.
Usually I just google for "spi wiring" or "[device] connection" or similar terms, sometimes, adding ESP or Arduino refines the results.
On the hardware tab of ESPEasy you can find the pins used for the common SPI signals, and there you also have to enable the SPI interface (and reboot the ESP after initially enabling it). The pin(s) used for CS (device select or similar terms), RST (reset) and other connections are configured per device, as they aren't shared with other SPI devices (if multiple are connected).

melkati, 2021-03-04:
I tried many combinations but can't find a good one.
With many of them the EPS8266 doesn't boot.
After each change I test with these commands (from the tools page):
epdcmd,clear
epd,txtfull,0,0,HelloWorld

None of them does anything on the display.

That is why I'm asking for help. I'm out of ideas.

TD-er, 2021-03-04:
I just looked into the [schematics of an ESP8266 e-paper board sold by Waveshare](https://www.waveshare.com/w/upload/7/7e/ESP8266_e-paper.pdf)

[image]

And also the suggested pins mentioned in the source code:
https://github.com/letscontrolit/ESPEasy/blob/aef8f4d9b13de7885a3e01cc394dad3bf817f37d/src/_P096_eInk.ino#L107-L121

You mentioned you had troubles booting the ESP.
That's probably because you used GPIO-15  (NodeMCU notation: D8)
If you're using GPIO-15 you must probably use a PNP transistor since that pin must remain pulled low (!!) during boot.
See the documentation for the Nextion display where this problem also may occur: [ESPEasy Nextion](https://espeasy.readthedocs.io/en/latest/Plugin/P075.html#hardware-serial-wiring)

melkati, 2021-03-05:
Thank you! Will try with this information.