letscontrolit/ESPEasy #1050: Problems with %systime% since a few hours

micropet, 2018-03-10:
With the current mega version, a rule with %systime% will no longer be executed.
It still worked until this morning.
Did you change that?

It is this part:

on green do
if %systime%> 08:00:00 and %systime% <20:00:00
NeoPixelAll, 0,80,0,0 // green during the day
else
NeoPixelAll, 0,8,0,0 // green evening
endif
endon

Now at 23:20 NeoPixelAll, 0.80,0,0 is executed.

TD-er, 2018-03-10:
It shouldn't be changed.
I did add something to support `%systime_am%`
Hope I didn't introduce any bug with that.

micropet, 2018-03-10:
I think so. Since the last update a few hours ago, it's not working anymore.

TD-er, 2018-03-10:
Time is correct? (NTP reachable)

Do you by any chance have this rule in another file or rule-set?

micropet, 2018-03-10:
yes, see my 1. post

TD-er, 2018-03-10:
Maybe I overlooked it (twice), but I cannot see anything in your post about using not the first ruleset.

micropet, 2018-03-10:
[image]

micropet, 2018-03-10:
Hm, my Rules are the same since a week or so. I flashed only a new Version.

TD-er, 2018-03-10:
The thing is, something has been changed by @mvdbro to speed up handling of rules.
That was not done by regular pull-requests, but just commits.
One of these commits was about not checking all files, but assuming they are once checked.
I just packed them all into one merge to v2.0 branch, which you can see here: https://github.com/letscontrolit/ESPEasy/commit/2682b0bd2febff501d783ded72f485860cd8b0e4

micropet, 2018-03-10:
Huch!

TD-er, 2018-03-10:
One of the things changed, which may cause this:
https://github.com/letscontrolit/ESPEasy/blob/ff656593ec018fb378f64dc824bd4a8d8a48d496/src/Misc.ino#L1999-L2004
The check is added. It used to be parsed always.
So you could remove the check and only leave this:
```C++
line = parseTemplate(line, line.length());
line.trim();
```

micropet, 2018-03-10:
No, it does not work.

16175 : ACT : event gruen
16185 : EVENT: gruen
16347 : EVENT: MH-Z19#MHZ19Temp=22.20
16449 : EVENT: MH-Z19#U=0.00
20065 : EVENT: Clock#Time=Sat,23:54
36960 : WD : Uptime 1 ConnectFailures 0 FreeMem 12344

107640 : EVENT: MH-Z19#PPM=521.00
107681 : ACT : NeoPixelAll,0,80,0,0

TD-er, 2018-03-10:
Ah, found a bug (not sure if it is your bug)
Look for the (1-off) error:
https://github.com/letscontrolit/ESPEasy/blob/ff656593ec018fb378f64dc824bd4a8d8a48d496/src/Misc.ino#L1893-L1908

https://github.com/letscontrolit/ESPEasy/blob/ff656593ec018fb378f64dc824bd4a8d8a48d496/src/Misc.ino#L1924-L1934

So again, what rule file are you using? (filenames at the SPIFF storage)

micropet, 2018-03-10:
Rules Set 1 with:

`// ----- ESP-206 Wo
On System#Boot do    //When the ESP boots, do
	Publish %sysname%/IP,%ip%
	timerSet,1,60      //Set Timer 1 for the next event in 60 seconds
	// timerSet,2,20      //Set Timer 2 for the next event in 20 seconds
endon

On Rules#Timer=1 do  //When Timer1 expires, do
	Publish %sysname%/IP,%ip%
	Publish %sysname%/Time,%systime%
	Publish %sysname%/Uptime,%uptime%
	Publish %sysname%/RSSI,%rssi%
	timerSet,1,60       //Resets the Timer 1 for another 60 seconds
endon
// ----- 

on Motion#Pir>0 do  // nur tagsueber anzeigen
     if %systime% > 08:00:00 and %systime% < 20:00:00 	
          gpio,14,1  // Led2 pink ON
     endif	
endon

on Motion#Pir=0 do
     gpio,14,0  // Led2 pink OFF
endon
// ----- 

On MH-Z19#PPM<700 do
    NeoPixelAll,0,80,0,0  // gruen
endon
	 
// ----- 
on MH-Z19#PPM do
	if [MH-Z19#PPM]<=700
		event gruen
	endif
	if [MH-Z19#PPM]>=700 and [MH-Z19#PPM]<=1200
		event gelb
	endif
	if [MH-Z19#PPM]>=1200
		event rot
	endif	
endon

on gruen	do
	if %systime% > 08:00:00 and %systime% < 20:00:00  
		NeoPixelAll,0,80,0,0  // gruen tagsueber
	else
		NeoPixelAll,0,8,0,0   // gruen abends
	endif
endon

on gelb do
	if %systime% > 08:00:00 and %systime% < 20:00:00  	
		NeoPixelAll,90,50,0,0 // gelb tagsueber
	else 
		NeoPixelAll,8,4,0,0 
	endif
endon

on rot do
	if %systime% > 08:00:00 and %systime% < 20:00:00   	
		NeoPixelAll,80,0,0,0 // rot tagsueber
	else 
		NeoPixelAll,8,0,0,0 
	endif 
endon	
// ----------`

micropet, 2018-03-10:
[image]

TD-er, 2018-03-10:
The motion detector in your rules also should not work at this moment. Does it?

micropet, 2018-03-10:
motion detector sends via mqtt, but the led is off. That should be so

So some of the rules work.

TD-er, 2018-03-10:
Just to make sure, you did remove the first check "if (match || !codeblock) {" and the matching "}", I suggested and it still doesn't work?

micropet, 2018-03-10:
I think i have it:

         int comment = line.indexOf("//");
          if (comment > 0)
            line = line.substring(0, comment);

            // only parse [xxx#yyy] if we have a matching ruleblock or need to eval the "on" (no codeBlock)
            // This to avoid waisting CPU time...
            line = parseTemplate(line, line.length());
            line.trim();

          String lineOrg = line; // store original line for future use
          line.toLowerCase(); // convert all to lower case to make checks easier

TD-er, 2018-03-10:
I think that's the part I was talking about.
You did remove the if statement and now it works?

micropet, 2018-03-10:
yes, i removed
no, it works not.

TD-er, 2018-03-10:
I think we have to look into it tomorrow with a fresh look.

micropet, 2018-03-10:
Yes, good. I also wanted to go to bed.
Good night.

micropet, 2018-03-10:
I have just flashed the version from 25.02.2018.
Everything works.

TD-er, 2018-03-10:
Hmm, would be great if some later versions also work, because a lot has changed the last 2 weeks.

Just to be sure, you're running Mega branch?

micropet, 2018-03-10:
Yes, I always have the Mega Branch,

Something does not seem to be OK yet.
The LEDs are dark and red at the moment, but every 30 seconds they flash green for a few milliseconds.

I will check everything again tomorrow.

TD-er, 2018-03-10:
Tomorrow past 8 am, so all seems to be fine ;)

micropet, 2018-03-10:
Yes

micropet, 2018-03-11:
Gijs, I have to apologize !!

I typed the rule again by hand.
Now everything works again.

There must have disturbed a space or a TAB.
Although I do not understand it anyway. I had not changed anything in the rules for over a week.

Peter

TD-er, 2018-03-11:
OK, nice to hear :)
Maybe we should check for all kind of whitespaces and add some redundant checks to it.

You have a dump here of the rules, so maybe tomorrow do a compare?
Just curious what it could have been.

mvdbro, 2018-03-11:
The 1-off bug is fixed, although it did not stop rules1 from processing anyway. I also added debugging on serial output (because the large volume only serial) when level is set to DEBUG_DEV. Then it will show the rules as they are fed into the MATCH block, so after parsing templates and system variables. It also shows the status on some booleans that will aid in debugging when needed.

micropet, 2018-03-11:
I've packed up the working and non-functioning rule. Look pretty much the same.

I'll test the current version of you right away.
[Rules.zip](https://github.com/letscontrolit/ESPEasy/files/1800207/Rules.zip)

uzi18, 2018-03-11:
During testing rules conditions,  my friend have found a bug with Tabs and whitespaces.
Rules were interpreted incorectly. 

It was introduced by...  Editing rules in external editor and next pasted to espeasy.

uzi18, 2018-03-11:
Maybe replace tabs to spaces in lines will resolve this.

micropet, 2018-03-11:
Good to know.
I write my programs and the rules with Notepad ++ and then paste them.

micropet, 2018-03-12:
So, everything works for me again as desired.
Thanks for your help.

closed 2018-03-12 by micropet

uzi18, 2018-03-12:
You changed something in rules or there was a patch?

uzi18, 2018-03-12:
duplicate of #1065 ?