letscontrolit/ESPEasy #1054: sleep time over 255 seconds

kickbutts, 2018-03-11:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Set the sleep time. In the config menu.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
Tell us what should happen?
It should be possible to set a sleep time longer than 255 seconds.

### Actual behavior
Tell us what happens instead?
The input field (page config, filed name: Sleep awake time) does not allow a value greater 255.
Maybe its just an html form problem or a wrong setting.

### System configuration

Hardware:
D1 mini

Software or git version:
git version v2.0-20180311

TD-er, 2018-03-11:
The value that is limited to 255 is the "awake" time between sleeps.
The other timer ("Sleep Delay") is the sleeptime, which is maxed to 4294 seconds (limited by hardware) to ~1.2h

Budman1758, 2018-03-11:
Still waiting for support for an external RTC such as the  DS3231.  This particular module would be great. Seems to me it would solve a lot of deep sleep time issues.  Programmable timers for milliseconds to days and all in between.  

https://www.ebay.com/itm/DS3231-AT24C32-IIC-Precision-Real-Time-Clock-RTC-Memory-Module-for-Arduino/121936579559?epid=10014451982&hash=item1c63fc8be7:g:7V8AAOSwdJ9aOFnD

kickbutts, 2018-03-11:
Oh i see. My fault. Did not look close enough. The versions before were different.
Maybe you should think of renaming the fields to prevent errors like mine.

TD-er, 2018-03-11:
Yep naming is one thing we should definitely look into.
But it's also the hardest part of programming, finding proper names.

Micha-he, 2018-03-12:
Maybe: Sleep duration & Sleep delay !?

TD-er, 2018-03-12:
Sleep duration and Awake time?
I guess "Sleep delay" is too confusing given the current naming.

kickbutts, 2018-03-12:
**Sleep duration** sounds fine to me. 
I am not sure what the other value is doing. Is it keeping the ESP awake for the entered time? Then i would suggest: **Awake time between sleeps** ( a bit long but self explaining).

TD-er, 2018-03-12:
It has changed very recently.
First it was just a checkbox enable/disable sleep. (stored as a 0 or  not 0 in a byte)
I changed it into "0 = disabled" and otherwise "awake time in seconds"
Actually it is more like "maximum time awake before going to sleep again", since you can put it to sleep earlier using rules.
Good luck finding a description of max. 20'ish characters ;)

zerog2k, 2018-08-20:
The wiki doesn't mention this "sleep awake time" and I too am confused by what the purpose of this is.
https://www.letscontrolit.com/wiki/index.php/SleepMode

TD-er, 2018-08-20:
We must add better documentation on this.

TD-er, 2018-10-12:
@adrianmihalko You can set the "awake" time to match some time in which all sensors can collect their data and send it.
You can also call the sleep from the rules, which I think is more reliable, since that can be done event based.

TD-er, 2018-10-21:
It looks like starting [Core 2.4.1](https://github.com/esp8266/Arduino/pull/4234) there is support for more than 2^32 usec (71 minutes)
The deep sleep now supports 64-bit ints, but the max you can use is determined by `deepSleepMax()` and should not set any longer or else it will not wake up any more.
The max is now somewhere between 3:25h and 3:50h.

I will have a look at how this can be implemented.

Wiki591, 2018-12-09:
Any progress on increasing the max sleep time?

Maybe the possibility to configure the 108th byte of init parameter and use of deep_sleep_set_option(0) stated in the documentation of espressif for low power solutions may increase the running time on battery?

TD-er, 2019-10-27:
This has been implemented. The max. sleep time will be computed and can be set.

closed 2019-10-27 by TD-er