letscontrolit/ESPEasy #1055: Enable VCC by default

kickbutts, 2018-03-11:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)

### Expected behavior
Tell us what should happen?
Enable VCC by default. For battery powered devices it is necessary to have a look at the battery. Why not enable VCC by default?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
WeMos Mini D1

Software or git version:
v2.0-20180311

ShardanX, 2018-03-11:
IMHO not a good idea. Many devices (e.g. GP2Y10, LDR, all MQ-Sensors without breakout) need the analog input, so VCC by default would break these devices.
If i remember right there was a suggestion about using a checkbox in config, I think it's the better choice if possible.

TD-er, 2018-03-11:
A checkbox is not possible, since you have to decide it at compile-time.
Well the main problem is you have to set it "outside" a function, before setup.
So you cannot read it from a settings file and thus has to be set at compile time.
But there are builds with VCC set.

Budman1758, 2018-03-11:
Well... Pretty easy to setup a jumper on your circuit board project to connect it externally. Only disadvantage is a bit more work building your board and then you need a task to use it.

kickbutts, 2018-03-11:
VCC builds are only available for PUYA. Maybe the build system can be configured to produce every at every build an VCC Version and an non VCC Version of all different types. On the other hand it would double the amount of files. Maybe this is confusing.

TD-er, 2018-03-11:
Hmm, that's not what's being setup in the PlatformIO.ini file.
So maybe we should look into that, because it was meant to have more VCC builds than only for the Puya.

There is currently a pending change in the build files to make it easier to define a specific build and then we should look into creating more versions with VCC enabled.

Budman1758, 2018-03-11:
There is a "Test" 4096  non-PUYAbuild in the zip every night

Grovkillen, 2018-03-12:
@Budman1758 is correct. In the nightly for both v2.0 and mega branches there's two VCC enabled bin files.

```
ESP_Easy_[branch_name]-20180312_test_ESP8266_4096_VCC.bin
ESP_Easy_[branch_name]-20180312_dev_ESP8266PUYA_1024_VCC.bin
```

I close this issue... please reopen if I misunderstood something.

closed 2018-03-12 by Grovkillen

kickbutts, 2018-03-12:
Maybe it is possible to build a 4096 developer version with VCC? This might be the most flexible version to work with.

Grovkillen, 2018-03-12:
Already is..? Have you looked in the latest releases?

Grovkillen, 2018-03-12:
Ah, sorry... Dev!