letscontrolit/ESPEasy #1056: 8574 boot issues on 20180311

giig1967g, 2018-03-11:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

TD-er, 2018-03-11:
This is not descriptive.
Please explain what's wrong

giig1967g, 2018-03-11:
After updating from 20180217 I found the following problem:
some relay connected to a 8574 unit turn on during boot.

In detail: 
On V2, during boot, the first (and sometimes the second) relay turn on during boot. In my system#boot section I have set them to off, so after few seconds they are turned off.
On MEGA, during boot, the 1st, 2nd, 3rd and 4th relay turn on during boot. In my system#boot section I have set them to off, so after few seconds they are turned off.

Tested with Wemos Mini (8266) and Mini Lite (8285).
PCF8574 is set on addresses: 65-72.

giig1967g, 2018-03-11:
(sorry took time to write the description)

giig1967g, 2018-03-11:
Actually tested again and with 8266 on MEGA and V2 the relays on boot up are the first 5.

giig1967g, 2018-03-11:
Also on 8285 on MEGA and V2 the relays on boot up are the first 5.
It looks like it's not always the same behaviour.
But the problem is that the relays gets on on boot.

uzi18, 2018-03-16:
Please show us Your Rules

giig1967g, 2018-03-16:
Hi, please close as the problem was related to the VCC.
I was using VCC=3.3V and there were problems.
With VCC=5V there were no more problems.

Maybe it should be mentioned in the wiki. Could help others :)

giig1967g, 2018-03-16:
I was referring to the PCF 8574 power.
Powering the PCF 8574 at 3.3V was generating issues.
Powering the PCF 8574 at 5.0V was OK.