letscontrolit/ESPEasy #1057: AM PM display issue.  CAUTION!! Nitpicking ahead!

Budman1758, 2018-03-11:
I REALLY hate to be a nitpick buuuuuut.....

Currently the display is like this   01:24:22pm   or am
Conventional (asfaik)  is like this.  1:24 PM  or AM
AM or PM should be capitol letters and have a space after the last number.  I have never seen the leading zero on any AM\PM display.

Also, IMHO displaying the seconds are not necessary unless they update dynamically.
Like I said, I hate to nitpick but methinks it would look much cleaner that way.

TD-er, 2018-03-11:
Would you be so kind to also have a look at the foot/inch notation?
Then we could fix that in the same run.
My choice for am/pm in lower case was due to space on the OLED Framed display with 1 line. (and because my first digital watch had am/pm really small in a corner of the screen)

Budman1758, 2018-03-12:
Not too sure how to check that.  I don't use anything that would display it.  I believe convention would be    12' 3''   OR     12ft 3in
If it were my choice I would probably use the former.

As for the space on an OLED line if you remove the seconds then space is not an issue.  Almost all AM\PM displays I see anywhere leave out the seconds.  Especially since they do not update every second.  Or perhaps you could provide for both and let the user decide.

Grovkillen, 2018-03-12:
Okay here's the decided (imperial) syntax

for length (current syntax=no change):
```
X' XX"
```

for time:
```
X:XX PM (for hours 0...9)
X:XX AM (for hours 0...9)
XX:XX PM (for hours 10...12)
XX:XX AM (for hours 10...12)
```

TD-er, 2018-03-12:
Since this topic was about nitpicking...
AM/PM notation doesn't have "0"-hours. :)

Grovkillen, 2018-03-12:
Fixed! :slightly_smiling_face:

uzi18, 2018-03-12:
OffTopic: @Budman1758 did You tested fixed pcf8574 plugin?

Budman1758, 2018-03-12:
@uzi18   Have not tried it yet.  I will be soon though.

TD-er, 2018-03-15:
@Grovkillen Can you add it to the documentation?
See description PR #1092 
Then it can be closed.

Grovkillen, 2018-03-16:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

Done! Will make a nicer table for this though, later today.

Grovkillen, 2018-03-16:
Made a NICE table ;)

closed 2018-03-16 by Grovkillen

Budman1758, 2018-03-21:
@uzi18 PCF8574 is now working as it should!!  NICE!!!