letscontrolit/ESPEasy #1059: Support for TM1638

twilight0x00, 2018-03-11:
I eould like to ask to for support gor TM1638. It is a very useful board with 8 segment display as well as 8 buttons and 8 leds. Since tm1637 is already supported, maybe it is not a big issue to support the x38 as well. I have code samples and a header file and can provide them if necessary.

papperone, 2018-03-30:
It's a board I've seen but never purchased (I can't think of a real life setup which can fit it) thus the IC is not supported in my plugin...
as per my experience coding without the hardware is not a nice way as I'll be impossible to test it and due to the IC being different the initial iteration to compile/test are needed; sorry!

twilight0x00, 2018-03-31:
Thank you for the answer. The real useful scenario would be to drive the 8
digits and may be the LEDs. I also cant think of a usefull scenario for the
buttons. If the hardware is the only obstacle, i will be happy to donate 1
board. Best regards.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er