letscontrolit/ESPEasy #1064: Add json library

s0170071, 2018-03-12:
As a preparation for JSON based configuration and JSON generation in general I suggest to decide which json library to use.  An add it to the repo for a start. 
I looked a bit and https://github.com/bblanchon/ArduinoJson seems to be "the" choice.

TD-er, 2018-03-12:
Why not cereal?
That can generate serializers for almost any type, import and export to json or binary.

s0170071, 2018-03-12:
Nice. Now I remember, you mentioned that one before. 
I like it. 
Will you add it then ?

TD-er, 2018-03-12:
Yeah sure, I will look at it ASAP :)
For completeness, the link to the project [Cereal](http://uscilab.github.io/cereal/)

uzi18, 2018-03-12:
What about ram/flash usage comparison?

TD-er, 2018-03-12:
RAM usage is certainly part of the review.
Flash usage is hard to say, since Cereal at least will generate some code to create the serializers. The big advantage is that it is not a user that has to maintain that code. (prone to errors)
At least they claim to be 'small' and 'efficient' and 'fast'.