letscontrolit/ESPEasy #1065: Rules doesnt fire anymore

DittelHome, 2018-03-12:
### Steps to reproduce
How can we trigger this problem?
If you use more  triggers insider rules, some trigger doesnt fire.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Yes it does.

### Expected behavior
Tell us what should happen?
I see the event in syslog, but action is missed.

### Actual behavior
Tell us what happens instead?
Action is missing !

### System configuration
**This is my rule:**
```
on System#Boot do // System IPL
   timerSet,1,1   //Set Timer 1 (Time, that LCD needs for init)
   lcdcmd,off     //off goes always
endon

on Rules#Timer=1 do        // Timer1 expired (LCD init time)
     lcdcmd,clear
     lcd,1,1,Waiting...    // line1 ==> Out
     lcd,2,1,Waiting...    // line2 ==> In/Date
     pwm,5,250
     lcdcmd,on
endon

on MQTT_IMPORT#out_hum do      
       lcd,1,1,Out T:[MQTT_IMPORT#out_temp] H:[MQTT_IMPORT#out_hum]   ende
       timerSet,5,3
endon 

on T&H#in_hum do
     lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    ende
     timerSet,2,3
endon

on Clock#Time=All,**:*0 do   // Check alle 10 Minuten
   If %systime% < 06:00:00 or  %systime% > 17:00:00
    pwm,5,2
   else
    pwm,5,250
   endif
endon

on Rules#Timer=2  do  // Timer2 expired, start flip-flop line 2
      lcd,2,1,%sysweekday_s% %sysday%-%sysmonth%-%sysyear%     ende
     timerSet,3,3                              
endon

on Rules#Timer=3  do  // Timer3 expired, goto timer 2 
     lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    ende
     timerSet,2,3     // flip-flop every 3 seconds
endon

on Rules#Timer=4  do  // Timer4 expired, start flip-flop line 1
      lcd,1,1,WS  [MQTT_IMPORT#ws_max] km/h max       ende
     timerSet,5,3     // flip-flop every 3 seconds
endon

on Rules#Timer=5  do  // Timer5 expired, goto timer 4 
     lcd,1,1,Out T:[MQTT_IMPORT#out_temp] H:[MQTT_IMPORT#out_hum]   ende
     timerSet,4,3     // flip-flop every 3 seconds
endon
```
**Here is the syslog:**
```
228934 : ACT : lcd,1,1,Out T:8.9 H:90 ende
228963 : ACT : timerSet,4,3
232699 : EVENT: Rules#Timer=4
232752 : ACT : lcd,1,1,WS 0 km/h max ende
232784 : ACT : timerSet,5,3
236265 : DHT : Temperature: 21.10
236265 : DHT : Humidity: 38.90
236267 : EVENT: T&H#in_temp=21.10
236295 : EVENT: T&H#in_hum=38.90
236325 : EVENT: Rules#Timer=5
236383 : ACT : lcd,1,1,Out T:8.9 H:90 ende
236412 : ACT : timerSet,4,3
240310 : EVENT: Rules#Timer=4
240364 : ACT : lcd,1,1,WS 0 km/h max ende
240393 : ACT : timerSet,5,3
243622 : EVENT: Rules#Timer=5
243682 : ACT : lcd,1,1,Out T:8.9 H:90 ende
243711 : ACT : timerSet,4,3
```
After the EVENT: T&H#in_temp=21.10, the rule engine should send the command: lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    ende, and then call the "timerSet,2,3", but nothing happen. The rule MQTT_IMPORT#out_hum works fine. Until mega 2018-03-09 all was working fine, after 03-09 the error occurs.

Hardware:
Wemos D1 mini

Software or git version:
With mega 2018-03-09 all is working fine, until 03.11 the problem occurs.

TD-er, 2018-03-12:
Can you check if any tab exists in your rules text?

DittelHome, 2018-03-12:
I have checked now, no tab inside th rules1.txt. (Check with notepad++)
After flashing the mega version from 2018-03-09, the rules is working as aspected. This tells me, the rules1.txt must be ok, right ?

TD-er, 2018-03-12:
Well there have been some changes and there was another issue this weekend, which looks like some "invisible" characters may cause these issues.

TD-er, 2018-03-12:
https://github.com/letscontrolit/ESPEasy/issues/1050

DittelHome, 2018-03-12:
This is the syslog from with ESPEasy_mega-20180309:
```
55052 : ACT : lcd,2,1,Mon 12-03-2018 ende
55083 : ACT : timerSet,3,3
55437 : EVENT: Rules#Timer=5
55973 : ACT : lcd,1,1,Out T:8.9 H:91 ende
56004 : ACT : timerSet,4,3
58254 : EVENT: Rules#Timer=3
58622 : ACT : lcd,2,1,In T:21.1 H:39 ende
58654 : ACT : timerSet,2,3
59261 : DHT : Temperature: 21.10
59261 : DHT : Humidity: 38.80
59262 : EVENT: T&H#in_temp=21.10
59887 : EVENT: T&H#in_hum=38.80
60133 : ACT : lcd,2,1,In T: H:39 ende
60164 : ACT : timerSet,2,3
60559 : EVENT: Rules#Timer=4
61006 : ACT : lcd,1,1,WS 0 km/h max ende
61037 : ACT : timerSet,5,3
63291 : EVENT: Rules#Timer=2
63565 : ACT : lcd,2,1,Mon 12-03-2018 ende
63596 : ACT : timerSet,3,3
```
This lines are missing 
59262 : EVENT: T&H#in_temp=21.10
59887 : EVENT: T&H#in_hum=38.80
**60133 : ACT : lcd,2,1,In T: H:39 ende**

uzi18, 2018-03-12:
Please download rules files from flash and attach here as zip archive

DittelHome, 2018-03-12:
[rules1.zip](https://github.com/letscontrolit/ESPEasy/files/1801681/rules1.zip)

Grovkillen, 2018-03-12:
Could the many spaces after `timerSet,3,3                              ` be a problem?

uzi18, 2018-03-12:
Or UTF8?

DittelHome, 2018-03-12:
> Could the many spaces after timerSet,3,3 be a problem?

Good idea, but not the solution..

> Or UTF8?

Ok, i think UTF without BOM should it be ?
I have tried it, also, but without success

DittelHome, 2018-03-12:
I got it...
After deletion of:
on T&H#in_hum do
     lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    ende
     timerSet,2,3
endon
and retyping it agian, it works.
Many thanks for your  help, and sorry for your effort !!

It is dangerous if someone tries to restore the rules1.txt with copy and paste, while using for example notepad++.

Grovkillen, 2018-03-12:
The OP issue is closed but the main problem here is that many users will do the exact same thing: edit it in a external text editor and copy past into the web interface. We need a clean up procedure upon save of a rule set.

TD-er, 2018-03-12:
Yay! we have a 'UX' category bug :)

Budman1758, 2018-03-12:
I just read through this twice and I still don't know exactly what caused the problem.  I use notepad++ also and have had no problem that I can attribute to it.  What was the exact issue?

Grovkillen, 2018-03-12:
Some non-printable letters/characters are being added to the chunk of text. Probably have to do with your national settings, keyboard or similar. Hard to tell...

TD-er, 2018-03-12:
The exact cause is yet unknown.
But one of the more 'tricky' parts in the code is that it expects a "\n\r" (ASCII code 10 + 13, or "\r\n", I always forget) line ending. That's a typical "Windows" line ending, while other OS'es typically use "\n" (ASCII code 10)
And perhaps also ASCII code 8 (tab) may disturb the parser.
Then there's also the possibility some UTF-8 encoding is present, which is always tricky to determine at what position a character starts in a string, since a single UTF-8 character may use 1..6 bytes to encode.
I guess the parse special characters function should always be called, since that one will also fix a few of these. It is now in a if-scope.

DittelHome, 2018-03-12:
After more test's, i found out, that the problem occurs, if you insert a blank after a endon:
[image]

**Removing blank after endon is the solution...
Inserting with notepad++ cuases no problem.**

A endon keyword with appended blank will be ignored !

uzi18, 2018-03-12:
see #1067

mvdbro, 2018-03-12:
TD wrote :"But one of the more 'tricky' parts in the code is that it expects a "\n\r" (ASCII code 10 + 13, or "\r\n"

This is incorrect, look at the code:
```
      if (data == 10)    // if line complete, parse this rule
      {
```

But the real issue is here:
```
          if (match || !codeBlock) {
            // only parse [xxx#yyy] if we have a matching ruleblock or need to eval the "on" (no codeBlock)
            // This to avoid waisting CPU time...
            line = parseTemplate(line, line.length());
            line.trim();
          }
```
The line.trim() is no longer performed on all lines, causing the trailing space to fail. Needs to be fixed.

uzi18, 2018-03-12:
\r\n are ok. fix is inf my PR #1067

mvdbro, 2018-03-12:
Why the label "Urgent" ? This is the dev branch right?
People should not use this in production.

TD-er, 2018-03-12:
To get the number of support requests down to a minimum

uzi18, 2018-03-12:
ok, integrate my patch, and will preprare next PR  ;P

closed 2018-03-12 by mvdbro