letscontrolit/ESPEasy #1068: Underscore in hostname is not allowed

bsiege, 2018-03-12:
hostname is generated from unitname_unitnumber
although it is sometime used, strictly according RFC952 it is NOT allowed.

I found this because dynamic updates failed:
sonoff3_3.int.mynet.com/A: bad owner name (check-names)

Simplest workaround: replace with minus-sign or make it configurable

mattlward, 2018-03-13:
I use the underscore in all of my ESP unit names, format ESP_(locaton).  Works fine for me on the latest build of the non-mega fork.

bsiege, 2018-03-13:
Yes, it works for you. For me too.. with workarounds. But this report is for developers who can understand a potential problem in professional and RFC-compliant DHCP/DNS environments.

mattlward, 2018-03-14:
Good point, we get set in our ways.  I am a network engineer at a large pubic university with around 70000 concurrent wireless users, but we assign all host names on wireless to prevent issues and I had not even given RFC compliance a thought.  My ESP's at work connect and post their IP address with MQTT and I always access them by IP address since name and address can and do change from time to time.

bsiege, 2018-03-21:
For me the workaround in bind 9.11 is adding `check-names warn;` to the dynamic zone.

Grovkillen, 2018-04-06:
Closing time?

TD-er, 2018-04-06:
Nope, I still have to replace _all_ underscores in the name, not only the _ between name and unit number.

Grovkillen, 2018-04-06:
Aha, as in "even if the user put underscore the host name will show hyphen (-)"?

Grovkillen, 2018-04-11:
I really think this is fixed.
[image]
[image]

closed 2018-04-11 by Grovkillen