letscontrolit/ESPEasy #1070: Manual reboot- how to find the cause

DittelHome, 2018-03-12:
### Steps to reproduce
How can we trigger this problem?
After a while (couple of hours/days) you should see manual reboots under /tools/info

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Yes, it  happen again..

### Expected behavior
Tell us what should happen?
The uptime shoulld rise at least to  several days.

### Actual behavior
Tell us what happens instead?
The esp8266 gets manual reboots. How can a debug the cause for that ?
I have already checked power many times (added capacitor, different power supplies etc.).
There are only a lcd dispaly and one dht22 connected. In addition a MQQT-Import is defined.

### System configuration

Hardware:
Wemos D1 mini

Software or git version:
actual mega git

TD-er, 2018-03-12:
Is the ESP connected to a PC. In other words, can something communicate over the USB => serial port?

I noticed some programs on my PC will always cause a reset of the connected ESP board.
One of them is Cura (for 3D prints)

DittelHome, 2018-03-12:
> Is the ESP connected to a PC. In other words, can something communicate over the USB => serial port?

The ESP ist installed in the garden shed,

Grovkillen, 2018-03-12:
Is it powered through the USB port or VIN (marked 5V on WeMos D1 mini)?

DittelHome, 2018-03-12:
I should go a little further, i have two esp in action, one in a wall clock, the other in the garden shed. Both have the problem that they get sporadic manaul reboots. 
Both are powered via a DC/DC converter to the VIN pin. The DC/DC converter is able to give 2 amperes. In addition i have connected some capitors (200 nF and 470uF))  to the 3.3 V and 5V pin's. I have already tried different DC/DC convertes, without success, Therefore i dont believe that the converter is the cause. (I know all threads about reboots says, power and again more power)  
The garden shed is powered through a big car battery (with solarpanel/charger). 
I am not sure, but when the ESP gets undervoltage it should get a cold boot right ?

TD-er, 2018-03-12:
On the forum I read a few times about "Wemos D1 mini" clones which gave some issues.
Especially with sleep mode, WiFi and/or power consumption.
I couldn't find the right image right now, but there is a photo indicating the visual differences between them.

DittelHome, 2018-03-21:
Problem solved !
After replacing my ESP8266 to new one, the uptime is now growing. I am not sure, if the replacement was neccassary, maybe a reset helps too. Problem determation in such cases are difficult. Sending syslog data to a syslog server is not much useful, i saw only a reboot.

closed 2018-03-21 by DittelHome

TD-er, 2018-03-21:
You could also try to completely erase the flash and try to see if that helps.
It is still possible you had a faulty one, which is not unlikely given the price of these things.